zephyrproject-rtos/zephyr #23441: RFC: API change: Add I2C bus recovery API

henrikbrixandersen, 2020-03-13:
## Introduction

Add API for recovering an I2C bus. This API can be used to recover from situations where the I2C master and one or more I2C slaves are out of synchronisation.

### Problem description

Synchronisation loss can occur e.g. if the I2C master was reset in the middle of an I2C transaction (watchdog timeout, for instance) or if a noise pulse was induced on the SCL line.

### Proposed change

Add a new I2C system call `int i2c_recover_bus(struct device *dev);` for allowing users to initiate an I2C Bus recovery. Extend the I2C controller driver API with methods for performing the bus recovery - either by using a generic, GPIO based approach (likely requires pinmux support) or by I2C controller specific functionality (some I2C controllers support bus recovery).

## Detailed RFC

The API proposed is inspired by the Linux kernel counterpart (see e.g. https://elixir.bootlin.com/linux/v5.5.9/source/include/linux/i2c.h#L595) but simplified for use in Zephyr (only support GPIO-based or I2C controller specific implementation, initially no support for controllers, which can control their SCL/SDA lines directly).

### Proposed change (Detailed)

TBD

### Dependencies

The GPIO-based approach to recovering an I2C bus requires changing the pinmux setup from the driver (in the prepare/unprepare functions) at runtime.

### Concerns and Unresolved Questions

None so far.

## Alternatives

Custom, board specific I2C Bus recovery routines.

henrikbrixandersen, 2020-03-13:
There are many sources for more background information on the I2C synchronisation issue. See e.g. https://community.nxp.com/thread/316813#comment-882281 which explains it very well.

carlescufi, 2020-03-17:
API meeting:

@MaureenHelm:
- Should we wait until we have a better overview of pinmux before we introduce this API, since you need to switch the pins from the I2C peripheral to the GPIO one in order to bitbang a reset of the bus.
@pabigot 
- If we could get the pins and DT flags from the I2C driver this could be done collaboratively with the I2C driver (acquire gpio, release gpio).

henrikbrixandersen, 2020-05-04:
PR  #23442 with a proposed API for I2C bus recovery is ready for review.

@carlescufi Can you please include it in the next API meeting agenda?

pabigot, 2020-05-06:
Following up on a comment in the API meeting yesterday that adding this was not a change to a stable API:

* It is backwards compatible, in that 2.2 code will work in 2.3 (because it won't be using the new API).
* It is not forwards compatible, in that 2.3 code may not work in 2.2 (where the new API doesn't exist)

So IMO it is a change to a stable API, but it is not an incompatible change.  The process does not describe what's to be done for these changes, but I think it should be mentioned in "API Changes" in the 2.3 release notes, as we don't maintain a version number for APIs that could be incremented per semver.

commit 71b89305 (2020-05-08): drivers: i2c: add I2C bus recovery API

Add API for recovering an I2C bus. This API can be used to recover
from situations where the I2C master and one or more I2C slaves are
out of synchronization (e.g. if the I2C master was reset in the middle
of an I2C transaction or if a noise pulse was induced on the SCL
line).

Fixes #23441.
closed 2020-05-08 by carlescufi