zephyrproject-rtos/zephyr #23443: esp32 needs i2c_transfer call  to turn on the display

aleksmarcelo, 2020-03-13:
**Describe the bug**
Hello folks,
i am learning how Zephyr works making a lot of tests in different boards and i think i got some sort of bug or incomplete code. The esp32 MCU is needing a call to the i2c_transfer function to turn on the OLED display. The samecode without the i2c_transfer works in a nrf51822.

**To Reproduce**
wiring a oled to the i2c pins and using the code:
```c
void display_init(void)
{
    int           sizebuf = 128 * 64 / 8;
    char          buffer[sizebuf];
    unsigned char datas[6];
    void *        cmd;

    
    dev = device_get_binding(DISPLAY_DRIVER);
    printk("\n\n\n\n\nDevice display=%d \n", dev);

    i2c_dev = device_get_binding(I2C0_DRIVER);
    printk("I2C driver= %d \n", i2c_dev);

    uint8_t setup[] = {//SETTINGS FOR ADAFRUIT 132x32 DISPLAY, PINS SOUTH & RIBBON RIGHT
                       00,
                       0xAE,  //DISPLAY OFF
                       0xD5,
                       0x80,  //DISPLAY CLOCK, SET TO RESET VALUE
                       0xA8,
                       31,  //MULTIPLEX RATIO, SET TO RESET VALUE (32 rows)
                       0xD3,
                       0x00,  //DISPLAY OFFSET, SET TO RESET VALUE (Desloca o)
                       0x40,  //DISPLAY START REGISTER 0b000000
                       0xA1,  //PINS SOUTH
                       0xC8,  //PINS SOUTH
                       0xDA,
                       0x02,  //SET DEFAULT COM PIN REMAP
                       0x81,
                       0x7F,  //SET CONTRAST LEVEL
                       0xDB,
                       0x30,  //SET DESELECT LEVEL VCOM MAX
                       0xA4,  //OUTPUT FOLLOWS RAM
                       0xA6,  //SET NORMAL DISPLAY (NOT INVERSE)
                       0x20,
                       0x00,  //SET HORIZONTAL ADDRESSING MODE
                       0x21,
                       0x00,
                       0x7F,  //SET COLUMN ADDRESS, START, END
                       0x22,
                       0x00,
                       0x07,  //SET PAGE ADDRESS, START, END
                       0xD9,
                       0x22,  //SET PRE-CHARGE PERIOD?
                       //-----
                       0xA8,
                       63,  // set multiplex: 0x1F for 128x32, 0x3F for 128x64
                       0xDA,
                       0x12,  // set compins - use 0x12 for 128x64
                       0xA4,  // display all on
                       0xA6,  // display normal (non-inverted)
                       //-----
                       0x8D,
                       0x14,   //CHARGE PUMP, ENABLE
                       0xAF};  //DISPLAY ON

    bool    extVcc   = false;

    i2c_write(i2c_dev, setup, sizeof(setup), 0x3c);

 #ifdef CONFIG_BOARD_ESP32
    struct i2c_msg msgs;
    u8_t           dst;
    msgs.buf   = &dst;
    msgs.len   = 0U;
    msgs.flags = I2C_MSG_WRITE | I2C_MSG_STOP;
    int i = 0;
    i2c_transfer(dev, &msgs, 1, 0x3c);
 #endif

    for (int g = 0; g < 6; g++)
    {
        oledSendCommand3(SSD1306_SET_COLUMN_ADDR, 0, 127);
        oledSendCommand3(SSD1306_SET_PAGE_ADDR, 0, 7);
        for (int c = 0; c < 8 * 8 - 1; c++)
        {
            cmd =  "\x40\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00";  
            i2c_write(i2c_dev, cmd, 17, 0x3c);
        }
        oledSendCommand3(SSD1306_SET_COLUMN_ADDR, 0, 127);
        oledSendCommand3(SSD1306_SET_PAGE_ADDR, 0, 7);
        for (int c = 0; c < 8 * 8; c++)
        {
            cmd =  "\x40\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff";   
            i2c_write(i2c_dev, cmd, 17, 0x3c);
            //  k_sleep(K_MSEC(5));
        }
    }
}
```

**Expected behavior**
Work on both boards without particular code.

**Impact**
Time discovering specific boards workarounds.

**Screenshots or console output**
None

**Environment (please complete the following information):**
Linux delpix 5.5.0-050500-generic #202001262030 SMP Mon Jan 27 01:33:36 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux

-- Zephyr version: 2.2.0-rc3
-- Found PythonInterp: /usr/bin/python3.6 (found suitable version "3.6.9", minimum required is "3.6")
-- Selected BOARD nrf51_ble400
-- Found west: /home/aleks/.local/bin/west (found suitable version "0.7.2", minimum required is "0.6.0")

**Additional context**
None

carlescufi, 2020-03-13:
@aleksmarcelo would you mind sending a Pull Request to fix this?

aleksmarcelo, 2020-03-13:
Hello @carlescufi, i am involved in a huge system and  all the staff is analyzing to make the MCU development using Zephyr as framework, but  we are at the beginning, so i will need some time to have all the knowledge involved .
 
We have read all the documentation on site, but it is not sufficient to understand all Zephyr internals.

Would you recommend any other good document ?

carlescufi, 2020-03-23:
> Hello @carlescufi, i am involved in a huge system and all the staff is analyzing to verify if Zephyr fits as framework to develop the MCU code, but we are at the beginning, so i will need some time to have all the knowledge involved .
> 
> We have read all the documentation on site, but it is not sufficient to understand all Zephyr internals.
> 
> Could you recommend any other good document ?

Hi @aleksmarcelo.
Unfortunately today that is all that there is as far as I know. Please do feel free to send an email to the mailing list or join our Slack and ask questions in both forums!

closed 2020-07-23 by github-actions[bot]