zephyrproject-rtos/zephyr #23449: "make clean" doesn't clean a lot of generated files

andrewboie, 2020-03-13:
Steps to reproduce:

```
cd tests/kernel/mem_protect/userspace
mkdir out
cd out
cmake -DBOARD=qemu_x86 ..
make -j20
make clean
find zephyr
```

I would not expect to see generated files still remaining, for example:

```
zephyr/zephyr.map
zephyr/app_smem_unaligned_prebuilt.map
zephyr/include
zephyr/include/generated
zephyr/include/generated/snippets-rom-start.ld
zephyr/include/generated/devicetree_unfixed.h
zephyr/include/generated/uart_ns16550_port_1.h
zephyr/include/generated/version.h
zephyr/include/generated/app_smem.ld
zephyr/include/generated/snippets-rodata.ld
zephyr/include/generated/snippets-noinit.ld
zephyr/include/generated/uart_ns16550_port_0.h
zephyr/include/generated/syscalls
zephyr/include/generated/syscalls/pwm.h
zephyr/include/generated/syscalls/zsock_getsockopt_mrsh.c
zephyr/include/generated/syscalls/net_if_ipv6_addr_lookup_by_index_mrsh.c
zephyr/include/generated/syscalls/i2c_configure_mrsh.c
zephyr/include/generated/syscalls/espi_write_flash_mrsh.c
zephyr/include/generated/syscalls/gpio_get_pending_int_mrsh.c
zephyr/include/generated/syscalls/k_sem_take_mrsh.c
zephyr/include/generated/syscalls/zsock_shutdown_mrsh.c
zephyr/include/generated/syscalls/adc.h
zephyr/include/generated/syscalls/espi_config_mrsh.c
zephyr/include/generated/syscalls/counter_set_top_value_mrsh.c
zephyr/include/generated/syscalls/net_if_ipv6_addr_add_by_index_mrsh.c
zephyr/include/generated/syscalls/net_eth_get_ptp_clock_by_index_mrsh.c
zephyr/include/generated/syscalls/k_timer_start_mrsh.c
zephyr/include/generated/syscalls/k_sem_reset_mrsh.c
zephyr/include/generated/syscalls/zsock_sendto_mrsh.c
zephyr/include/generated/syscalls/z_log_string_from_user_mrsh.c
zephyr/include/generated/syscalls/k_float_disable_mrsh.c
zephyr/include/generated/syscalls/k_msgq_put_mrsh.c
zephyr/include/generated/syscalls/zsock_getsockname_mrsh.c
zephyr/include/generated/syscalls/log_ctrl.h
zephyr/include/generated/syscalls/sensor_channel_get_mrsh.c
zephyr/include/generated/syscalls/k_uptime_get_mrsh.c
zephyr/include/generated/syscalls/hwinfo.h
zephyr/include/generated/syscalls/hwinfo_get_device_id_mrsh.c
zephyr/include/generated/syscalls/k_msgq_peek_mrsh.c
zephyr/include/generated/syscalls/missing_syscall_mrsh.c
zephyr/include/generated/syscalls/uart_poll_out_mrsh.c
zephyr/include/generated/syscalls/mutex.h
zephyr/include/generated/syscalls/k_pipe_alloc_init_mrsh.c
zephyr/include/generated/syscalls/atomic_set_mrsh.c
zephyr/include/generated/syscalls/gpio_enable_callback_mrsh.c
zephyr/include/generated/syscalls/k_msgq_get_attrs_mrsh.c
zephyr/include/generated/syscalls/uart_err_check_mrsh.c
zephyr/include/generated/syscalls/uart_drv_cmd_mrsh.c
zephyr/include/generated/syscalls/k_object_release_mrsh.c
zephyr/include/generated/syscalls/gpio_port_set_masked_raw_mrsh.c
zephyr/include/generated/syscalls/sensor_sample_fetch_chan_mrsh.c
zephyr/include/generated/syscalls/k_poll_signal_reset_mrsh.c
zephyr/include/generated/syscalls/k_thread_priority_set_mrsh.c
zephyr/include/generated/syscalls/flash_get_page_info_by_offs_mrsh.c
zephyr/include/generated/syscalls/atomic_add_mrsh.c
[... 237 lines cut ...]
zephyr/include/generated/syscalls/z_zephyr_write_stdout_mrsh.c
zephyr/include/generated/syscalls/peci_disable_mrsh.c
zephyr/include/generated/syscalls/k_timer_status_sync_mrsh.c
zephyr/include/generated/syscalls/k_poll_mrsh.c
zephyr/include/generated/syscalls/espi_flash_erase_mrsh.c
zephyr/include/generated/syscalls/test_syscall.h
zephyr/include/generated/syscalls/i2s_buf_read_mrsh.c
zephyr/include/generated/syscalls/libc-hooks.h
zephyr/include/generated/syscalls/k_wakeup_mrsh.c
zephyr/include/generated/syscalls/k_futex_wait_mrsh.c
zephyr/include/generated/syscalls/zsock_connect_mrsh.c
zephyr/include/generated/syscalls/uart_tx_abort_mrsh.c
zephyr/include/generated/syscalls/uart_poll_in_mrsh.c
zephyr/include/generated/syscalls/i2c_slave_driver_unregister_mrsh.c
zephyr/include/generated/devicetree.conf
zephyr/include/generated/app_data_alignment.ld
zephyr/include/generated/snippets-rwdata.ld
zephyr/include/generated/devicetree_fixups.h
zephyr/include/generated/autoconf.h
zephyr/include/generated/snippets-sections.ld
zephyr/include/generated/otype-to-size.h
zephyr/include/generated/snippets-ram-sections.ld
```

Found when debugging #23429 which makes the problem worse for reasons not yet clear to me.

SebastianBoe, 2020-03-16:
This is probably due to us not using BYPRODUCTS when we should.

If a script generates a file, and we don't tell CMake about it, then CMake won't know that it should delete it.

Until all cases are fixed, users can consider using "make pristine", but note that this deletes the Makefile as well, which is not always desired.

carlescufi, 2021-02-09:
@tejlmand do you think this is worth pursuing? Maybe documenting the limitations might be another option.

tejlmand, 2021-02-09:
> @tejlmand do you think this is worth pursuing? Maybe documenting the limitations might be another option.

Yes. It is always worth to ensure that byproducts (and dependencies) are set up correctly everywhere in the build syste.

I have briefly looked into this, and the most tricky part is the autogenerated files where we don't know which files are being generated by the scripts, and hence hard to cleanup using byproducts, but from CMake 3.15 there is a new flag that can be used for directories that hold such generated files.
(There is also Makefile support for CMake <3.15, but as we generally recommend Ninja, I haven't looked more into this)

The biggest part of this is going manually through the code to find all occurrences, so trade off between effort and gain means this has been low on prioritization list.with severity.

tejlmand, 2021-02-09:
@carlescufi made a first round here: #32140

PR #32140 (2021-02-14): cmake: improved handling of output and byproducts in CMake
Fixes: #23449

This commit adds additional OUTPUT and BYPRODUCTS to custom command
and targets in the Zephyr build system.

This ensures that files produced during the build will be removed again
when invoking ninja clean / make clean.

The generated syscalls headers include folder is added to the syscall
target using ADDITIONAL_CLEAN_FILES property.
However, this property is new in CMake 3.15, so will not work when using
older CMake with ninja.

For CMake versions <=3.15 the ADDITIONAL_MAKE_CLEAN_FILES property is
used. However, this  [... cut]
commit c4c79f5f (2021-02-14): cmake: improved handling of output and byproducts in CMake

Fixes: #23449

This commit adds additional OUTPUT and BYPRODUCTS to custom command
and targets in the Zephyr build system.

This ensures that files produced during the build will be removed again
when invoking ninja clean / make clean.

The generated syscalls headers include folder is added to the syscall
target using ADDITIONAL_CLEAN_FILES property.
However, this property is new in CMake 3.15, so will not work when using
older CMake with ninja.

For CMake versions <=3.15 the ADDITIONAL_MAKE_CLEAN_FILES property is
used. However, this [... cut]
commit 116425cc (2021-02-19): cmake: improved handling of output and byproducts in CMake

Fixes: #23449

This commit adds additional OUTPUT and BYPRODUCTS to custom command
and targets in the Zephyr build system.

This ensures that files produced during the build will be removed again
when invoking ninja clean / make clean.

The generated syscalls headers include folder is added to the syscall
target using ADDITIONAL_CLEAN_FILES property.
However, this property is new in CMake 3.15, so will not work when using
older CMake with ninja.

For CMake versions <=3.15 the ADDITIONAL_MAKE_CLEAN_FILES property is
used. However, this [... cut]
closed 2021-02-14 by nashif