zephyrproject-rtos/zephyr #23452: Allow applications to define their own ZephyrBinaryRunners

mbolivar, 2020-03-13:
It would be nice for applications to be able to define their own runners. This would make it possible for west flash and friends to work with an application without board-specific hacks to pull the runners in.

Currently, only boards can configure runners. This is a bit silly for cases like network protocols that can run a Zephyr binary somewhere else.

Originally requested at https://github.com/zephyrproject-rtos/zephyr/pull/23336#discussion_r389555199.

mbolivar, 2020-03-13:
I never meant to assign you. Sorry. Not sure how that happened.

mbolivar-nordic, 2020-03-17:
I've switched my work-related account to @mbolivar-nordic (https://github.com/zephyrproject-rtos/zephyr/pull/23530).