zephyrproject-rtos/zephyr #23465: Zephyr BLE Authentication - ZEBRA

deangereaux, 2020-03-14:
# ZEBRA - Zephyr Bluetooth Authentication

March 2020

Bluetooth has become the defacto standard for connecting devices over short distances.  Everything from 
clocks, speakers, cars, to small low power IoT devices use Bluetooth Low Energy (BLE).  BLE 
is by far the most popular choice for configuring and managing IoT devices.  Bluetooth’s 
popularity has also made it an attractive hacking target.

ZEBRA is a proposal to add secure Bluetooth authentication in an easy to use manner for developers. 

### Problem description

While the Bluetooth protocol has become more secure and robust, one of the biggest challenges is 
authenticating a Bluetooth connection.  For consumer products, authenticating a Bluetooth 
enabled device using a pairing code is sufficient in most use cases.  For commercial IoT 
applications, such as building lighting controls, using a pairing code is very insecure and most 
IoT devices do not have a screen or the expectation of a human in the loop.  

“Authentication” in this context means securely verifying the identity of the connecting device, be 
it a mobile application or another IoT device.  The design of Bluetooth is intended to allow 
interoperability between devices, however for many devices this is unacceptable.  Should a 
Bluetooth enabled glucose monitoring system allow any mobile application to connect? 
No. How does a low-cost, not Internet connected, lighting control device authenticate a mobile 
 application connecting via Bluetooth?  

### Proposed change

The goal of ZEBRA (Zephyr Bluetooth Authentication), is to provide a simple, open source
mechanism to add Bluetooth authentication for Zephyr based devices using the Zephyr 
KConfig system.  Ease of use is critical. The developer can select an authentication method and the Bluetooth layer 
(GATT or L2CAP), the necessary code is then included into the project. 

Initially two authentication methods are proposed, DTLS and Challenge-Response.  Additional
authentication methods can be added including custom authentication such as a RSA hardware 
token based authentication method.  The authentication is performed at the Bluetooth application 
level versus the stack itself.  This has both advantages and disadvantages, discussed later in this proposal.

## Detailed RFC

The ZEBRA design consists of an API layer, KConfig interface, and the underlying authentication methods.  
>An initial implementation can be found here: https://github.com/GoldenBitsSoftware/zephyr    see: `feature/ble_authentication` branch.  BLE Authentication source files are located here:  subsystem/bluetooth/services/auth_*.c.  Sample source files are located here: samples/bluetooth/authenticated_connection

##### Authentication API

The Authentication API is designed to abstract away the authentication method and BLE interface 
(GATT or L2CAP).  The calling application configures the method and BLE interface, 
starts the authentication process and monitors results via a status callback.  The API is also 
designed to handle multiple concurrent authentication processes, for example If device is acting 
as a Central and Peripheral.  An example of the API used is shown in the following code snippet.

``` 
static authenticate_conn central_auth_conn;

void auth_status(struct authenticate_conn *auth_conn, auth_status_t status, void *context)
{
    if(status == AUTH_STATUS_SUCCESSFUL) {
        printk(“Authentication Successful.\n”);
    } else {
        printk(“Authentication status: %s\n”, auth_svc_get_status_str(status));
    }
}

/* BLE connection callback */
void connected(struct bt_conn *conn, u8_t err) 
{
   /* start authentication */
   central_auth_conn.conn = conn;
   auth_svc_start(&central_auth_conn);
}
[... 111 lines cut ...]
_Google Weave Certificate compression._  Weave is an open source code solution to compressing X.509 certificates up
to 40%.  This does increase the handshake speed and reduces the amount of resources needed.  Weave compressed 
certificates will be a developer selectable option.  
See [Google Weave Digital Certs Spec](https://openweave.io/reference/specifications/Protocol_Specification_Weave_Digital_Certificates.pdf)

## Alternatives

#### Comparison of Bluetooth Pairing and Authentication
Simply put, for commercial IoT applications pairing is not authentication. The Bluetooth version 5 specification
does define three methods of authentication (Bluetooth spec version 5, Vol 2, Part F, Section 4.2.9):  a) Numeric 
Comparison, b) Passkey, and c) Out of Band.  All of these methods require a display on the device and a 
human in the loop.  In all of these options, the user is  manually performing the authentication.  For 
commercial IoT applications, this is not an option. Imagine a commercial building where hundreds 
of light switches (the IoT devices) must be manually paired. Error prone and costly work.
 

Pairing establishes an encryption key for both parties (Central and Peripheral), which is different 
from validating the identity of the remote peer.  This is true for both Legacy and LE Secure pairing. While LE
Secure pairing uses an ECC keypair and Diffie-Hellman Elliptic Curve to protect against eavesdropping,
there is no capability to identify the remote peer.

#### GATT Authentication
GATT Authentication defined in the Bluetooth spec (Version 5, Vol 3, Part G, Section 8.1) is used to 
enable a per characteristic authentication using a shared key.  It does not authenticate a Bluetooth 
peer (Central or Peripheral).

aescolar, 2020-03-16:
CC: @jhedberg @carlescufi @Vudentz

carlescufi, 2020-03-16:
@deangereaux thanks for the submission!

Could you please include a small paragraph about why the existing [fast-pair spec](https://developers.google.com/nearby/fast-pair/spec) is not a viable alternative?

deangereaux, 2020-03-16:
@carlescufi   Added paragraph.  I view GFPS as another possible authentication method provided by ZEBRA.   Let me know if I missed any key points about GFPS.  Thanks!!

dleach02, 2020-03-16:
> However Why not use this shared key for BLE link layer encryption? Why not set this key into the Zephyr Bluetooth stack after the DTLS handshake

Not sure you need to complicate this as it seems the key establishment in BLE is okay. Since you are attempting to deal with the "I have the device" and "I'm authenticated" to use the device.

endian-benjamin, 2020-03-27:
Maybe I'm missing a crucial paragraph, but it seems to me that this has little to do with Bluetooth specifically? I mean, it seems you want to decouple DTLS from its traditional UDP/IP dependency, which seems like a noble goal to me, but this all sounds equally applicable to for example direct serial communication, or NFC, or the homebrew 433 MHz Footooth protocol I just made up. Right?

Why not just decouple it further - an easy-to-use, transport-agnostic authentication lib on one hand, and a Bluetooth-specific implementation on the other?

deangereaux, 2020-03-27:
The API is designed to be transport agnostic so yes in future versions there is no reason why ZEBRA shouldn’t be able to authenticate over a serial line or other transport.  The ZEBRA design allows for different authentication methods and potentially different transports by adding a different transport specific thread, serial for example.  However the transport and authentication are not cleanly decoupled as suggested, which has been previously suggested and is a very good idea.   I’ll take hard look at implementing this separation. 

The reason Bluetooth is the initial focus is simple, it is the most common use case.  This is actually why I started creating ZEBRA, I’ve been asked many times over the past 3 years to design and develop an authentication method for a phone connected device.  

Thanks for the feedback.

carlescufi, 2020-04-14:
API meeting:

Dean proposes: `subsys/auth/zebra` and `include/auth/zebra.h`

galak, 2020-04-16:
dev-review: (Apr 16) - suggestion is put it under lib.

Some comment about having this as a module:
* discussion is this really a feature we expect/want in the main repo (vs a module)
* some issues currently ci for modules today (vs main repo)
* where do we draw the line about would would be a module v main repo for zephyr project code.

deangereaux, 2020-04-16:
@pabigot   Regarding the current state of: https://github.com/GoldenBitsSoftware/zephyr.  Yes I need to update, rebase, etc.

deangereaux, 2020-07-07:
Current status.   
a) Moving code to lib/auth
b) Refactoring code to make library independent of transport (ie Bluetooth, Serial, etc).
c) Adding transport layer API.

pabigot, 2020-07-07:
API 2020-07-07: Moved from triage to in-progress.

deangereaux, 2020-08-05:
Current status:
a) The transport independent layer is in place. 
b) Adding serial (UART) transport example.
c) Added code to handle message (dis)assembly when sending a larger message over a transport with smaller MTU.
d) Created `samples/authentication` directory to contain all authentication samples.
e) Current work in progress can be found here:  https://github.com/GoldenBitsSoftware/zephyr/tree/feature/auth_lib

deangereaux, 2020-10-16:
I've re-worked the majority of the code, still more to do but getting close to formally creating a PR.  Here is the latest design document.  
[ZAUTH_Zephyr_9-2020.pdf](https://github.com/zephyrproject-rtos/zephyr/files/5392217/ZAUTH_Zephyr_9-2020.pdf)

Chandranshu9, 2020-12-22:
Please help me integrate Zephyr authentication- ZAUTH so that I can run sample bluetooth authentication code available.
Below is the error I am getting while doing build in Ubuntu terminal:

Parsing /home/chandranshu/zephyrproject/zephyr/Kconfig
Loaded configuration '/home/chandranshu/zephyrproject/zephyr/boards/arm/arduino_nano_33_iot/arduino_nano_33_iot_defconfig'
Merged configuration '/home/chandranshu/zephyrproject/zephyr/zephyr-feature-auth_lib/samples/authentication/bluetooth/central_auth/prj.conf'

error: HAS_NRFX (defined at modules/Kconfig.nordic:187) is assigned in a configuration file, but is
not directly user-configurable (has no prompt). It gets its value indirectly from other symbols. See
http://docs.zephyrproject.org/latest/reference/kconfig/CONFIG_HAS_NRFX.html and/or look up HAS_NRFX
in the menuconfig/guiconfig interface. The Application Development Primer, Setting Configuration
Values, and Kconfig - Tips and Best Practices sections of the manual might be helpful too.

CMake Error at /home/chandranshu/zephyrproject/zephyr/cmake/kconfig.cmake:251 (message):
  command failed with return code: 1
Call Stack (most recent call first):
  /home/chandranshu/zephyrproject/zephyr/cmake/app/boilerplate.cmake:608 (include)
  CMakeLists.txt:4 (include)

-- Configuring incomplete, errors occurred!

Chandranshu9, 2020-12-22:
If this is not the right place to ask this question then please tell me where to ask.

Please help me integrate Zephyr authentication- ZAUTH so that I can run sample bluetooth authentication code available.
Below is the error I am getting while doing build in Ubuntu terminal:

Parsing /home/chandranshu/zephyrproject/zephyr/Kconfig
Loaded configuration '/home/chandranshu/zephyrproject/zephyr/boards/arm/arduino_nano_33_iot/arduino_nano_33_iot_defconfig'
Merged configuration '/home/chandranshu/zephyrproject/zephyr/zephyr-feature-auth_lib/samples/authentication/bluetooth/central_auth/prj.conf'

error: HAS_NRFX (defined at modules/Kconfig.nordic:187) is assigned in a configuration file, but is
not directly user-configurable (has no prompt). It gets its value indirectly from other symbols. See
http://docs.zephyrproject.org/latest/reference/kconfig/CONFIG_HAS_NRFX.html and/or look up HAS_NRFX
in the menuconfig/guiconfig interface. The Application Development Primer, Setting Configuration
Values, and Kconfig - Tips and Best Practices sections of the manual might be helpful too.

CMake Error at /home/chandranshu/zephyrproject/zephyr/cmake/kconfig.cmake:251 (message):
command failed with return code: 1
Call Stack (most recent call first):
/home/chandranshu/zephyrproject/zephyr/cmake/app/boilerplate.cmake:608 (include)
CMakeLists.txt:4 (include)

-- Configuring incomplete, errors occurred!

deangereaux, 2020-12-22:
@Chandranshu9   `HAS_NRFX` is platform specific for the Nordic nRF52 SOCs and should not be in the `prj.conf` file.  Go ahead and remove and try to re-create.  Also I fixed a bug recently, the stack size of the main thread was too small.   You should also add `CONFIG_MAIN_STACK_SIZE=4096` to the proj.conf file.

ZAUTH is still a work in progress, so there will be bugs.  I'm hoping to finish and submit a formal PR by the end of December.

Chandranshu9, 2020-12-22:
> @Chandranshu9 `HAS_NRFX` is platform specific for the Nordic nRF52 SOCs and should not be in the `prj.conf` file. Go ahead and remove and try to re-create. Also I fixed a bug recently, the stack size of the main thread was too small. You should also add `CONFIG_MAIN_STACK_SIZE=4096` to the proj.conf file.
> 
> ZAUTH is still a work in progress, so there will be bugs. All of the initial development has been with the Nordic nRF52840 Dev kit (see: https://www.nordicsemi.com/Software-and-tools/Development-Kits/nRF52840-DK). I'm hoping to finish and submit a formal PR by the end of December.

Hello @deangereaux, Thanks for your quick response, Yes removing HAS_NRFX solved my issue, Now I am receiving some Kconfig warnings, The same warnings were also shown when I was running the test version of authentication. Please help me with this. I am eagerly waiting for the final version of your ZAUTH, it looks very promising.
Below is the error that I am receiving while building the code:-

Parsing /home/chandranshu/zephyrproject/zephyr/Kconfig
Loaded configuration '/home/chandranshu/zephyrproject/zephyr/boards/arm/arduino_nano_33_iot/arduino_nano_33_iot_defconfig'
Merged configuration '/home/chandranshu/zephyrproject/zephyr/zephyr-feature-auth_lib/samples/authentication/bluetooth/central_auth/prj.conf'

/home/chandranshu/zephyrproject/zephyr/zephyr-feature-auth_lib/samples/authentication/bluetooth/central_auth/prj.conf:6: warning: attempt to assign the value 'y' to the undefined symbol BT_GATT_DIS

/home/chandranshu/zephyrproject/zephyr/zephyr-feature-auth_lib/samples/authentication/bluetooth/central_auth/prj.conf:22: warning: attempt to assign the value 'y' to the undefined symbol AUTH_LIB

/home/chandranshu/zephyrproject/zephyr/zephyr-feature-auth_lib/samples/authentication/bluetooth/central_auth/prj.conf:23: warning: attempt to assign the value 'y' to the undefined symbol BT_XPORT

/home/chandranshu/zephyrproject/zephyr/zephyr-feature-auth_lib/samples/authentication/bluetooth/central_auth/prj.conf:24: warning: attempt to assign the value 'y' to the undefined symbol AUTH_CHALLENGE_RESPONSE

/home/chandranshu/zephyrproject/zephyr/zephyr-feature-auth_lib/samples/authentication/bluetooth/central_auth/prj.conf:27: warning: attempt to assign the value 'y' to the undefined symbol AUTH_CHALLENGE_RESPONSE

/home/chandranshu/zephyrproject/zephyr/zephyr-feature-auth_lib/samples/authentication/bluetooth/central_auth/prj.conf:31: warning: attempt to assign the value '3' to the undefined symbol AUTH_LOG_LEVEL

/home/chandranshu/zephyrproject/zephyr/zephyr-feature-auth_lib/samples/authentication/bluetooth/central_auth/prj.conf:55: warning: attempt to assign the value 'y' to the undefined symbol MBEDTLS_CIPHER_CBC_ENABLED

error: Aborting due to Kconfig warnings

CMake Error at /home/chandranshu/zephyrproject/zephyr/cmake/kconfig.cmake:251 (message):
  command failed with return code: 1
Call Stack (most recent call first):
  /home/chandranshu/zephyrproject/zephyr/cmake/app/boilerplate.cmake:608 (include)
  CMakeLists.txt:4 (include)

-- Configuring incomplete, errors occurred!

deangereaux, 2020-12-22:
@Chandranshu9   Not sure why you're getting these errors.  I'll take a look later this week, in the mean time if you figure out what's causing these config errors, please let me know.

Chandranshu9, 2020-12-22:
@deangereaux  This is because I am not able to integrate your ZAUTH with zephyrRTOS properly, Can you tell me how to integrate them both.
I downloaded ZAUTH and pasted it inside zephyr folder. 
Are there any changes that need to be made for running your ZAUTH library properly with zephyrOS?
I think if you can tell me how to install this properly then these issues will be solved.

deangereaux, 2020-12-22:
@Chandranshu9   Let's move this conversation off this RFC doc.  Please enter this info here as an issue: https://github.com/GoldenBitsSoftware/zephyr/issues

Chandranshu9, 2020-12-22:
> @Chandranshu9 Let's move this conversation off this RFC doc. Please enter this info here as an issue: https://github.com/GoldenBitsSoftware/zephyr/issues

@deangereaux Sure, I have moved the issue. Please reply there.

deangereaux, 2021-01-07:
I'm happy to announce, a PR has been submitted for ZAUTH
PR: https://github.com/zephyrproject-rtos/zephyr/pull/31136

closed 2021-03-31 by deangereaux