zephyrproject-rtos/zephyr #23468: bluetooth: host: Runtime HCI_LE_Create_Connection timeout

JordanYates, 2020-03-15:
**Is your enhancement proposal related to a problem? Please describe.**
I have a common GATT central application that is expected to connect to many different peripheral applications. Consider 2 applications, one which advertises at 10Hz, another at 1Hz.

A timeout of 500ms is likely sufficient for connecting to application A, while 5s may be desired for application B.
Waiting for 5 seconds for application A is a waste of time, as if you haven't connected in 500ms, you likely wont connect in 5 seconds (device has probably moved out of range).

Time spent initiating a connection that will never happen is time that you are not scanning for useful information from other devices.

**Describe the solution you'd like**
An additional interface `bt_conn_create_timeout` or similar, that updates the timeout for initiating connections.

**Describe alternatives you've considered**
Currently I am implementing this functionality by waiting on a semaphore that is given in the connection callback, where the wait duration is my desired timeout. Connection is then canceled through bt_conn_disconnect if the semaphore times out.

JordanYates, 2020-03-15:
I will have a go at implementing this

joerchan, 2020-03-16:
@JordanYates Have you considered to use a whitelist and `bt_conn_create_aute_le` for this?

JordanYates, 2020-03-16:
@joerchan I hadn't previously, but it does not really fit the model of what we are trying to achieve.
We aren't interested in connecting to ANY device running application A or B. There can easily be over 50 devices  within bluetooth range, none of which I want to connect to. At some point in time, in response to an external command or as a result of the data that one particular device has sent, I want to connect to a specific device. Using the whitelists appears to add overhead without solving my actual problem, which is receiving a timely response that I can't connect to a device. Sometimes there is a user (often me) in the loop of this process, making the timely response more important.

PR #23911 (2020-04-06): bluetooth: host: Dynamic bt_conn_le_create timeout
Extends the bt_conn_le_create_param struct to provide the option
to set a custom timeout for the initiation of the connection.

Timeout is not added as a parameter to the BT_CONN_LE_CREATE_PARAM
macro due to the expectation that CONFIG_BT_CREATE_CONN_TIMEOUT
will be the typical value that users will expect.

Fixes #23468
commit a038dc76 (2020-04-06): bluetooth: host: Dynamic bt_conn_le_create timeout

Extends the bt_conn_le_create_param struct to provide the option
to set a custom timeout for the initiation of the connection.

The logic for the default values of window_coded and interval_coded
were moved to conn.c in order to resolve all defaults for the
create_param struct at a single location.

Timeout is not added as a parameter to the BT_CONN_LE_CREATE_PARAM
macro due to the expectation that CONFIG_BT_CREATE_CONN_TIMEOUT
will be the typical value that users will expect.

Fixes #23468
closed 2020-04-06 by jhedberg