zephyrproject-rtos/zephyr #23474: tests/subsys/usb/device failed on reel_board.

chen-png, 2020-03-16:
**To Reproduce**
Steps to reproduce the behavior:
1. west build -b reel_board -p auto tests/subsys/usb/device/
2. west flash
3. See error

**Screenshots or console output**
*** Booting Zephyr OS build zephyr-v2.2.0-368-g22b9167acb52  ***
D: lock usb_enable_lock mutex
D: USB detected
D: HF clock start success (0)
D: USB Powered
D: HF clock stop success (0)
D: USB Removed
D: set cb, ep: 0x81
D: EP enable: 0x00
D: EP enable: 0x80
D: unlock usb_enable_lock mutex
Running test suite test_device

starting test - test_usb_dc_api_invalid
D: ep_write: ep 0x20, len 1
D: ep_read: ep 0x20, maxlen 1
D: USB detected
D: HF clock start success (0)
D: USB Powered
D: HF clock stop success (0)
D: USB Removed
PASS - test_usb_dc_api_invalid

starting test - test_usb_deconfig
PASS - test_usb_deconfig

**starting test - test_usb_disable**
D: USB detected
**ASSERTION FAIL [m_drv_state == NRFX_DRV_STATE_INITIALIZED] @** /home/wentongw/work/zephyrproject/modules/hal/nordic/nrfx/drivers/src/nrfx_usbd.c:1729
E: r0/a1:  0x00000004  r1/a2:  0x000006c1  r2/a3:  0x00000001
E: r3/a4:  0x0000c312 r12/ip:  0x00000020 r14/lr:  0x000070d7
E:  xpsr:  0x61000000
E: Faulting instruction address (r15/pc): 0x0000ad3a
E: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
E: Current thread: 0x200013d0 (workqueue)
E: Halting system

**Environment (please complete the following information):**
 - OS: fedora 28
 - Toolchain: zephyr-sdk-0.11.1
 - Commit ID: 22b9167acb

jfischer-phytec-iot, 2020-03-18:
I can not reproduce it with latest master (000a7327e1bab2823185854cd6b6589f3bc7c7e1). Please recheck.

chen-png, 2020-03-19:
same error with 000a732, is there any mistake i made?
 i just used this command "west build -b reel_board -p auto tests/subsys/usb/device/"

*** Booting Zephyr OS build zephyr-v2.2.0-454-g000a7327e1ba  ***
D: lock usb_enable_lock mutex
D: USB detected
D: HF clock start success (0)
D: USB Powered
D: HF clock stop success (0)
D: USB Removed
D: set cb, ep: 0x81
D: EP enable: 0x00
D: EP enable: 0x80
D: unlock usb_enable_lock mutex
Running test suite test_device
....

starting test - test_usb_disable
D: USB detected
ASSERTION FAIL [m_drv_state == NRFX_DRV_STATE_INITIALIZED] @ WEST_TOPDIR/modules/hal/nordic/nrfx/drivers/src/nrfx_usbd.c:1729
E: r0/a1:  0x00000004  r1/a2:  0x000006c1  r2/a3:  0x00000001
E: r3/a4:  0x0000c330 r12/ip:  0x00000020 r14/lr:  0x000070d7
E:  xpsr:  0x61000000
E: Faulting instruction address (r15/pc): 0x0000ad42
E: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
E: Current thread: 0x200013d0 (workqueue)
E: Halting system

nashif, 2020-03-31:
cant reproduce on master.

nashif, 2020-03-31:
well, it is sporadic...

carlescufi, 2020-05-13:
@jfischer-phytec-iot please try to reproduce on reel_board
@emob-nordic please try to reproduce on nRF52840 DK

emob-nordic, 2020-05-14:
Couldn't reproduce the issue. I am using zephyr-sdk 0.11.2. Built form f95aed10cc master.

carlescufi, 2020-05-20:
@jfischer-phytec-iot please try to reproduce and, if not reproduceable anymore, please close.

chen-png, 2020-05-20:
maybe we can close this, this issue hasn't occurred in daily test more than 2 weeks recently

jfischer-phytec-iot, 2020-07-10:
Can not reproduce, close

closed 2020-07-10 by jfischer-phytec-iot