zephyrproject-rtos/zephyr #23475: tests/kernel/gen_isr_table failed on iotdk board.

chen-png, 2020-03-16:
**To Reproduce**
Steps to reproduce the behavior:
1.  sanitycheck -p iotdk --device-testing --device-serial /dev/ttyUSB0 -T tests/kernel/gen_isr_table/
2. See error in handler.log file

**Screenshots or console output**
** Booting Zephyr OS build zephyr-v2.2.0-368-g22b9167acb52  ***
starting test - Test gen_isr_tables
IRQ configuration (total lines 95):
isr1 isr=0x20000604 irq=94
isr2 isr=0x20000644 irq=93
Checking _irq_vector_table entry 78 for irq 94
triggering irq 94
isr1 ran
Checking _irq_vector_table entry 77 for irq 93
triggering irq 93
isr2 ran
_sw_isr_table at location 0x800013e8
isr3 isr=0x20000574 irq=92 param=0xb01dface
Checking _sw_isr_table entry 76 for irq 92
triggering irq 92
isr3 ran with parameter 0xb01dface
isr4 isr=0x20000598 irq=91 param=0xca55e77e
Checking _sw_isr_table entry 75 for irq 91
triggering irq 91
isr4 ran with parameter 0xca55e77e
isr5 isr=0x200005bc irq=90 param=0xf0ccac1a
Checking _sw_isr_table entry 74 for irq 90
triggering irq 90
isr5 ran with parameter 0xf0ccac1a
**ASSERTION FAIL [!arch_irq_is_enabled(irq)] @ WEST_TOPDIR/zephyr/arch/common/sw_isr_common.c:19
        IRQ 89 is enabled**
E: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
E: Current thread: 0x800000a4 (main)
E: Halting system

**Environment (please complete the following information):**
 - OS: fedora 28
 - Toolchain: zephyr-sdk-0.11.1
 - Commit ID: 22b9167acb

abrodkin, 2020-04-07:
I may confirm we do see the same in our internal nightly verification.

vonhust, 2020-04-27:
The failure is caused by the conflict of irq used in this test and peripheral irq.  UART3 will use IRQ 89 and is enabled although it should not be.   

PR #23479 will fix this issue.

PR #24739 (2020-04-28): board: iotdk: disable unused peripherals in dts
By defaut, only one uart is used as console.
Disable unused peripherals in dts to avoid
conflicts (issue #23475), save memory footprint,
and reduce the boot time.

Fixes #23475
closed 2020-04-28 by MaureenHelm