zephyrproject-rtos/zephyr #23482: 2M PHY + DLE and timing calculations on an encrypted link are wrong

thoh-ot, 2020-03-16:
**Describe the bug**
We are seeing some issue regarding 2M PHY + DLE and timing calculations on an encrypted link.

We use DLE with 251 as maximum and have some internal test that tries to send 244 bytes via GATT with 7 bytes protocol overhead, so that becomes 251 bytes LL PDU Payload. When adding the 4 byte MIC, the LL PDU length becomes 255.

On the AIR this should become one LL packet, instead we are seeing two.

We see that when running 2M that struct lll_conn's max_tx_time is 1060 and when ull_conn_lll_max_tx_octets_get(struct lll_conn *lll) [subsys/bluetooth/controller/ll_sw/ull_conn.c] is called max_tx_octets becomes 1060/8 - 11 = 254 where it should be 255.

The correct max_tx_time should be 1064, with 4us extra for the missing byte.

We believe we have tracked the issue to PKT_US(octets, phy) [subsys/bluetooth/controller/ll_sw/ull_conn_internal.h] where the phy parameter is fixed to 1, when given to PREAMBLE_SIZE(phy) [subsys/bluetooth/controller/ll_sw/pdu.h].

Also we have noticed that in some calls to PKT_US a 0 is given as phy which doesn't make sense, so we guess it should be BIT(0).

**Expected behavior**
On the AIR the 244 bytes GATT operation should be one LL packet.

**Impact**
Throughput degradation.

**Environment**
Zephyr 2.2

aescolar, 2020-03-16:
CC @cvinayak @kruithofa

kruithofa, 2020-03-18:
Same bug is present in the legacy controller

PR #23557 (2020-03-18): Bluetooth: controller: split: correct timing calculation in PKT_US
A bug in the PKT_US resulted in wrong calculations for the 2M phy.
Fixes the bug, verified on EBQ.
Also adds some defines for improved readability.

Fixes #23482
commit 30e3205e (2020-03-18): Bluetooth: controller: split: correct timing calculation in PKT_US

A bug in the PKT_US resulted in wrong calculations for the 2M phy.
Fixes the bug, verified on EBQ.
Also adds some defines for improved readability.

Fixes #23482
PR #23558 (2020-04-09): Bluetooth: controller: legacy: Correct DLE time calculations
This PR fixes #23482. The preamble size for a 2M phy was incorrect.

There is a bug in calculation of time for the DLE procedure:
the preamble size is incorrect for the 2M phy

This is for the legacy code, see PR #23557 for the split controller
Fixes #23482
commit aafd5631 (2020-04-09): Bluetooth: controller: legacy: Correct DLE time calculations

This PR fixes #23482. The preamble size for a 2M phy was incorrect.

There is a bug in calculation of time for the DLE procedure:
the preamble size is incorrect for the 2M phy

Fixes #23482
closed 2020-03-18 by aescolar