zephyrproject-rtos/zephyr #23496: Issue building & flashing a hello world project on NRF52840

Abz23, 2020-03-16:
Hi,

I am currently using a 64 bit windows 10 laptop and I am trying to build a simple hello world sample code in the zephyr OS into my NRF52840 DK. I have followed all the steps in the get started user guide and I have made sure to install all the dependencies that are required.

My problem is that I am unable to build as I am always presented with the error below. 

I am currently on windows 10 pro version 1909.

Any guidance would be greatly appreciated. 

C:\Zephyr\zephyrproject\samples\hello_world\build>cmake -GNinja -DBOARD=nrf52840_pca10056 ..
-- Application: C:/Zephyr/zephyrproject/samples/hello_world
-- Zephyr version: 2.2.99
-- git describe failed: fatal: No names found, cannot describe anything.;
   BUILD_VERSION is left undefined
-- Board: nrf52840_pca10056
-- Found west: C:/Python38/Scripts/west.exe (found suitable version "0.6.0", minimum required is "0.6.0")
FATAL ERROR: can't load west manifest: no "manifest.path" config option is set

CMake Error at C:/Zephyr/zephyrproject/cmake/zephyr_module.cmake:45 (message):
  Traceback (most recent call last):

    File "C:/Zephyr/zephyrproject/scripts/zephyr_module.py", line 242, in <module>
      main()
    File "C:/Zephyr/zephyrproject/scripts/zephyr_module.py", line 203, in main
      raise subprocess.CalledProcessError(cmd=p.args,

  subprocess.CalledProcessError: Command '['C:/Python38/Scripts/west.exe',
  'list', '--format={posixpath}']' returned non-zero exit status 1.

Call Stack (most recent call first):
  C:/Zephyr/zephyrproject/cmake/app/boilerplate.cmake:463 (include)
  CMakeLists.txt:5 (include)

-- Configuring incomplete, errors occurred!

C:\Zephyr\zephyrproject\samples\hello_world\build>

carlescufi, 2020-03-16:
@Abz23 you seem to be running west outside of a git repository.
Please follow the exact steps in https://docs.zephyrproject.org/latest/getting_started/index.html.

Abz23, 2020-03-17:
Thanks for getting back to me so quickly. 

I have made sure to repeat all the steps in the getting started link you posted but still with no luck. 

Can you please inform me what am missing as I have tried reinstalling and updating west?

C:\Users\A A\zephyrproject\zephyr>west build -p auto -b nrf52840_pca10056 samples/basic/blinky
-- west build: making build dir C:\Users\A A\zephyrproject\zephyr\build pristine
-- west build: generating a build system
-- Application: C:/Users/A A/zephyrproject/zephyr/samples/basic/blinky
-- Zephyr version: 2.2.99
-- Found PythonInterp: C:/Python38/python.exe (found suitable version "3.8.2", minimum required is "3.6")
-- Board: nrf52840_pca10056
-- Found west: C:/Python38/Scripts/west.exe (found suitable version "0.7.2", minimum required is "0.6.0")
-- Found toolchain: gnuarmemb (C:/gnuarmemb)
-- Loading C:/Users/A A/zephyrproject/zephyr/boards/arm/nrf52840_pca10056/nrf52840_pca10056.dts as base
Devicetree header saved to 'C:/Users/A A/zephyrproject/zephyr/build/zephyr/include/generated/devicetree_unfixed.h'
CMake Error at C:/Users/A A/zephyrproject/zephyr/cmake/kconfig.cmake:160 (message):
  File not found: C:/Users/A
Call Stack (most recent call first):
  C:/Users/A A/zephyrproject/zephyr/cmake/app/boilerplate.cmake:466 (include)
  CMakeLists.txt:4 (include)

-- Configuring incomplete, errors occurred!
FATAL ERROR: command exited with status 1: 'C:\Program Files\CMake\bin\cmake.EXE' '-BC:\Users\A A\zephyrproject\zephyr\build' '-SC:\Users\A A\zephyrproject\zephyr\samples\basic\blinky' -GNinja -DBOARD=nrf52840_pca10056

Abz23, 2020-03-17:
I think the issue is to do with this error below, but I am not sure how to rectify it.

Many thanks 

CMake Error at CMakeLists.txt:5 (include):
  include could not find load file:

    /cmake/app/boilerplate.cmake

-- The C compiler identification is unknown
-- The CXX compiler identification is unknown
CMake Error at CMakeLists.txt:6 (project):
  No CMAKE_C_COMPILER could be found.

  Tell CMake where to find the compiler by setting either the environment
  variable "CC" or the CMake cache entry CMAKE_C_COMPILER to the full path to
  the compiler, or to the compiler name if it is in the PATH.

CMake Error at CMakeLists.txt:6 (project):
  No CMAKE_CXX_COMPILER could be found.

  Tell CMake where to find the compiler by setting either the environment
  variable "CXX" or the CMake cache entry CMAKE_CXX_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.

Abz23, 2020-03-17:
My problem is that I am unable to init west , I keep getting this error

Any guidance would be appreciated

C:\zephyr>west init
=== Initializing in c:\zephyr
--- Cloning manifest repository from https://github.com/zephyrproject-rtos/zephyr, rev. master
Initialized empty Git repository in C:/zephyr/.west/manifest-tmp/.git/
remote: Enumerating objects: 119, done.
remote: Counting objects: 100% (119/119), done.
remote: Compressing objects: 100% (91/91), done.
error: RPC failed; curl 18 transfer closed with outstanding read data remaining
fatal: the remote end hung up unexpectedly
fatal: early EOF
fatal: index-pack failed
FATAL ERROR: command exited with status 128: git fetch origin --tags -- master 'refs/heads/*:refs/remotes/origin/*'

closed 2020-03-17 by Abz23