zephyrproject-rtos/zephyr #23507: samples/subsys/shell/shell_module doesn't work on qemu_x86_64

andrewboie, 2020-03-16:
**Describe the bug**
When I run this sample under qemu_x86_64 I do not get a shell prompt.

**Expected behavior**
An interactive shell prompt should be available that accepts commands.

**Impact**
Not sure if this is a sample issue or the shell subsystem is broken in some way. Could be an issue with the x86_64 arch, or an SMP problem.

andrewboie, 2020-03-16:
I think this is an SMP problem, the sample works fine if I set CONFIG_MP_NUM_CPUS=1

jenmwms, 2020-04-13:
> the sample works fine if I set CONFIG_MP_NUM_CPUS=1

@andrewboie  Thanks for this! Can you please help me reproduce this? I added `CONFIG_MP_NUM_CPUS=1` to the `prj.conf` file in `samples/subsys/shell/shell_module` but it still hangs for me.

jenmwms, 2020-04-16:
> `CONFIG_MP_NUM_CPUS=1` to the `prj.conf` file in `samples/subsys/shell/shell_module` but it still hangs for me.

nvm - this works to reproduce the passing case. Not sure what happened earlier. Anyway, this issue is still on my radar.

jenmwms, 2020-04-16:
btw - I see this same behavior for `qemu_x86_64` `tests/drivers/uart/uart_basic_api/`. The `test_uart_fifo_fill` fails, but works fine if I set `CONFIG_MP_NUM_CPUS=1`

jenmwms, 2020-05-07:
update here - I rebased today and this issue still occurs. However, this issue appears to be resolved with cherry-pick of #23325.

jenmwms, 2020-05-11:
update here - I rebased today and this issue is resolved. Above referenced #23325 fixed it, from what I can tell. 

I tested today on master 7201c1d - No modification or cherry-pick.
 `west build -p -b qemu_x86_64 samples/subsys/shell/shell_module`

```
[jennife2@jennife2-desk1 zephyr]$ west build -t run
-- west build: running target run
[3/4] To exit from QEMU enter: 'CTRL+a, x'[QEMU] CPU: qemu64,+x2apic
qemu-system-x86_64: warning: TCG doesn't support requested feature: CPUID.01H:ECX.x2apic [bit 21]
qemu-system-x86_64: warning: TCG doesn't support requested feature: CPUID.01H:ECX.x2apic [bit 21]
SeaBIOS (version rel-1.12.1-0-ga5cab58-dirty-20200214_052440-f7294c49af13-zephyr
)
Booting from ROM..Optimal CONFIG_X86_MMU_PAGE_POOL_PAGES 8

uart:~$

```

closed 2020-05-11 by jenmwms