zephyrproject-rtos/zephyr #23520: JLink Thread-Aware Debugging (RTOS Plugin)

JordanYates, 2020-03-17:
SEGGER J-Link is supported as a flash runner for a substantial portion of boards supported by Zephyr (131 across nrfjprog and jlink). However there is currently no support for thread-aware debugging using the suite of debug tools that use JLink as a backend (arm-none-eabi-gdbserver, J-Link GDB Server).
Those using external SEGGER jlink based flash tools are forced to choose between using JLink as their backend (and not having thread-aware debugging) or using openocd, which requires overwriting drivers on windows and appears to have been largely superceded by pyocd (Unfortunately pyocd does not support SEGGER J-Link Probes).

This is a long way of saying that Zephyr should support thread-aware debugging using the J-Link suite of tools.

RTOS-Awareness is added to J-Link through writing a custom plugin using a proprietary SDK (https://www.segger.com/products/debug-probes/j-link/tools/j-link-gdb-server/thread-aware-debugging/)

Writing a custom plugin appears to be relatively simple from the example provided with the SDK, but the licensing terms are not great. In short, publishing the source of the plugin is prohibited without the express consent of SEGGER. Only the compiled .dll or .so files are allowed to be distributed. (I am unsure as to whether the license file is included in this agreement, otherwise I would attach it).

Assuming a plugin is written, by myself or others, what would be the appropriate method to distribute it such that it can be used. There doesn't appear to have been any resolution to #7516

pkral78, 2020-03-20:
I definitely look at this. J-Link is industry standard debug probe - it's no-brainer.

Quick examination of existing plugins (embOS, FreeRTOS, ChibiOS) shows simple API to implements:

```
RTOS_Init
RTOS_GetVersion
RTOS_GetSymbols
RTOS_GetNumThreads
RTOS_GetCurrentThreadId
RTOS_GetThreadId
RTOS_GetThreadDisplay
RTOS_GetThreadRegList
RTOS_SetThreadRegList
RTOS_UpdateThreads
RTOS_GetThreadReg
RTOS_SetThreadReg
```

Regarding licencing, it would be better to ask SEGGER for their PoV.

pkral78, 2020-03-20:
And of course, in meantime, there is also possible of using gdb python API to examine thread structures.

richey-v, 2020-04-06:
@pkral78 - could you provide a link to that gdb python information?

pkral78, 2020-04-06:
@richey-v You have to define new gdb command that examine kernel structures. There are many references on internet, search for _python gdb scripting_. Also setting `CONFIG_OPENOCD_SUPPORT` would be helpfull. Look at `openocd.c` in Zephyr tree.

MaureenHelm, 2020-04-06:
You can also have a look at zephyr thread-awareness implementations in [pyocd](https://github.com/mbedmicro/pyOCD/blob/master/pyocd/rtos/zephyr.py) and [openocd](https://github.com/zephyrproject-rtos/openocd/blob/zephyr-20200116/src/rtos/Zephyr.c). Both require setting `CONFIG_OPENOCD_SUPPORT=y`

vaussard, 2020-04-07:
I have a couple of commands in my .gdbinit to help when debugging with JLink. They are not polished (i.e. PC/LR/PSP is wrong for the currently running thread), but maybe this can be useful to others. I posted them here:

https://gist.github.com/vaussard/581d62032670f118947edb2b503a18c4

Compile with `CONFIG_THREAD_MONITOR` and `CONFIG_THREAD_STACK_INFO`. Then run `zephyr_list_threads` from gdb to show all the threads.

The JLink plugin is also on my list, but I never managed to find some time to do it. So I am highly interested if someone manages to do it.

carlescufi, 2020-04-14:
@JordanYates and @pkral78 if you were willing to write an implementation and test it with Zephyr, then I could send a request to Segger to include this in the J-Link Software and Documentation Pack download.

carlescufi, 2020-12-03:
According to Segger, they are working on it:
https://twitter.com/SEGGERMicro/status/1298893256323932161

manoj153, 2021-01-28:
> According to Segger, they are working on it:
> https://twitter.com/SEGGERMicro/status/1298893256323932161

Any new when it will available?

rettichschnidi, 2021-03-25:
> Ozone now ships with a JavaScript RTOS awareness plugin for Zephyr.

[As of version 3.22d, Ozone is able to deal with Zephyr thread!](https://www.segger.com/products/development-tools/ozone-j-link-debugger/)

To activate it, type `Project.SetOSPlugin("ZephyrPlugin_CM4")` into the console window.

JordanYates, 2021-04-28:
As of JLink 7.20, Zephyr is supported:
https://www.segger.com/downloads/jlink/ReleaseNotes_JLink.html

closed 2021-04-28 by JordanYates

rettichschnidi, 2021-05-03:
Anyone having any success running the JLink Zephyr RTOS Plugin?

I tried the following, and failed:
1. Load plugin:
    ```patch
    $ git diff boards/arm/nrf52840dk_nrf52840/board.cmake
    diff --git a/boards/arm/nrf52840dk_nrf52840/board.cmake b/boards/arm/nrf52840dk_nrf52840/board.cmake
    index ef822795fe..6fa47f220d 100644
    --- a/boards/arm/nrf52840dk_nrf52840/board.cmake
    +++ b/boards/arm/nrf52840dk_nrf52840/board.cmake
    @@ -1,6 +1,6 @@
     # SPDX-License-Identifier: Apache-2.0
     
    -board_runner_args(jlink "--device=nrf52" "--speed=4000")
    +board_runner_args(jlink "--device=nrf52" "--speed=4000" "-rtos GDBServer/RTOSPlugin_Zephyr")
     board_runner_args(pyocd "--target=nrf52840" "--frequency=4000000")
     include(${ZEPHYR_BASE}/boards/common/nrfjprog.board.cmake)
     include(${ZEPHYR_BASE}/boards/common/jlink.board.cmake)
    ```
    Same goes for `-rtos=/opt/SEGGER/JLink_V720/GDBServer/RTOSPlugin_Zephyr.so` and other variations I tried.
2. Ensuring the thread monitor support is enabled: 
    ```bash
    $ grep CONFIG_THREAD_MONITOR zephyr/.config 
    CONFIG_THREAD_MONITOR=y
    ```
3. `west flash` and `west debug`
4. No thread information:
    ```
    (gdb) break main
    Breakpoint 1 at 0x0: main. (2 locations)
    (gdb) c
    Continuing.
    
    Breakpoint 1, main () at /home/XXX/src/main.c:10
    (gdb) info threads 
      Id   Target Id         Frame
    * 1    Thread 57005      main () at /home/XXX/src/main.c:10
    ```

MaureenHelm, 2021-05-03:
Make sure to set `CONFIG_DEBUG_THREAD_INFO=y`. See also #34660