zephyrproject-rtos/zephyr #23528: k64f dts flash0/storage_partition 8KiB -> 64KiB

drandreas, 2020-03-17:
**Describe the bug**

[boards/arm/frdm_k64f/frdm_k64f.dts](../blob/master/boards/arm/frdm_k64f/frdm_k64f.dts) reads as Follows:
```
/*
 * The flash starting at 0x00010000 and ending at
 * 0x0001ffff (sectors 16-31) is reserved for use
 * by the application.
 */
storage_partition: partition@1e000 {
  label = "storage";
  reg = <0x0001e000 0x00002000>;
};
```

To my understanding 64KiB (0x00010000 to 0x0001ffff) should be available to the application. However dts only assigned 8 KiB (<0x0001e000 0x00002000>) as storage space.

I highly suspect that 
`reg = <0x0001e000 0x00002000>;`
should be changed to 
`reg = <0x00010000 0x00010000>;`

**Impact**
Uncertainty of beginners if the space can freely be used or is for some reason it is reserved by the mbuboot and therefore not partitioned here.

**Environment (please complete the following information):**
 - OS: Ubuntu 18.04
 - Toolchain Zephyr SDK
 - 41481f92f7412694f32a3306b692ca8b7c899baa

**Additional context**
I'am trying to create a product w/ mcuboot for fw update and a storage_partition for permanent data.

drandreas, 2020-03-18:
It looks like it was also propagated into k66 dts #22736

parthitce, 2020-03-18:
@drandreas Can this be modified in overlay.dts specific to the application's storage needs?

drandreas, 2020-03-18:
Yes, I've now simply added overlays to my app as follows:

boards/frdm_k64f.overlay
```
// Resize storage to 64KiB
/delete-node/ &storage_partition;

&flash0 {
	partitions {
		storage_partition: partition@10000 {
			label = "storage";
			reg = <0x00010000 0x00010000>;
		};
	};
};
```

boards/ip_k66f.overlay
```
// Resize storage to 1MiB
/delete-node/ &storage_partition;

&flash0 {
	partitions {
		storage_partition: partition@100000 {
			label = "storage";
			reg = <0x00100000 0x00100000>;
		};
	};
};
```

drandreas, 2020-03-18:
But it remains strange that the documentation and the default layout do not match.

carlescufi, 2020-04-07:
@drandreas please send a Pull Request changing the default .dts value and we can discuss it there please.

closed 2020-04-07 by carlescufi