zephyrproject-rtos/zephyr #23539: west flash --runner jlink returns KeyError: 'jlink'

GeorgeGkinis, 2020-03-17:
I try to flash my program to a **bare-bones** **nrf52832** module using an original **jlink edu** mini from segger.

Build succeeds :

**PS C:\FUN\NRF5\zephyrproject\zephyr>west build ..\..\ble_beacon_zephyr**
-- west build: generating a build system
-- Application: C:/FUN/NRF5/ble_beacon_zephyr
-- Zephyr version: 2.2.99
-- Found PythonInterp: C:/Python38/python.exe (found suitable version "3.8.2", minimum required is "3.6")
-- Board: nrf52832_mdk
-- Found west: C:/Python38/Scripts/west.exe (found suitable version "0.7.2", minimum required is "0.6.0")
-- Found toolchain: gnuarmemb (C:/FUN/NRF5/9-2019-q4-major)
Devicetree header saved to 'C:/FUN/NRF5/zephyrproject/zephyr/build/zephyr/include/generated/devicetree_unfixed.h'
Parsing C:/FUN/NRF5/zephyrproject/zephyr/Kconfig
Loaded configuration 'C:/FUN/NRF5/zephyrproject/zephyr/boards/arm/nrf52832_mdk/nrf52832_mdk_defconfig'
Merged configuration 'C:/FUN/NRF5/ble_beacon_zephyr/prj.conf'
Configuration saved to 'C:/FUN/NRF5/zephyrproject/zephyr/build/zephyr/.config'
Kconfig header saved to 'C:/FUN/NRF5/zephyrproject/zephyr/build/zephyr/include/generated/autoconf.h'
-- The C compiler identification is GNU 9.2.1
-- The CXX compiler identification is GNU 9.2.1
-- The ASM compiler identification is GNU
-- Found assembler: C:/FUN/NRF5/9-2019-q4-major/bin/arm-none-eabi-gcc.exe
-- Cache files will be written to: C:\Users\dbkggk\AppData\Local/.cache/zephyr
-- Configuring done
-- Generating done
-- Build files have been written to: C:/FUN/NRF5/zephyrproject/zephyr/build
-- west build: building application
[162/167] Linking C executable zephyr\zephyr_prebuilt.elf
Memory region         Used Size  Region Size  %age Used
           FLASH:       60461 B       512 KB     11.53%
            SRAM:       27992 B        64 KB     42.71%
        IDT_LIST:         120 B         2 KB      5.86%
[167/167] Linking C executable zephyr\zephyr.elf

Afterwards I do :  

 **PS C:\FUN\NRF5\zephyrproject\zephyr>west flash --runner jlink**  
-- west flash: rebuilding
ninja: no work to do.
-- west flash: using runner jlink
Traceback (most recent call last):
  File "c:\python38\lib\runpy.py", line 193, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "c:\python38\lib\runpy.py", line 86, in _run_code
  File "C:\Python38\Scripts\west.exe\__main__.py", line 9, in <module>
  File "c:\python38\lib\site-packages\west\app\main.py", line 768, in main
    app.run(argv or sys.argv[1:])
  File "c:\python38\lib\site-packages\west\app\main.py", line 106, in run
    self.run_command(argv)
  File "c:\python38\lib\site-packages\west\app\main.py", line 334, in run_command
    self.run_extension(args.command, argv)
  File "c:\python38\lib\site-packages\west\app\main.py", line 397, in run_extension
    command.run(args, unknown, self.topdir, manifest=self.manifest)
  File "c:\python38\lib\site-packages\west\commands.py", line 116, in run
    self.do_run(args, unknown)
  File "c:\fun\nrf5\zephyrproject\zephyr\scripts\west_commands\flash.py", line 30, in do_run
    do_run_common(self, my_args, runner_args)
  File "C:\FUN\NRF5\zephyrproject\zephyr\scripts/west_commands\run_common.py", line 168, in do_run_common
    runner_config['args'][runner_name] +
**KeyError: 'jlink'**

Using **nrfjprog** everything goes smoothly :  

PS C:\FUN\NRF5\zephyrproject\zephyr> nrfjprog --eraseall -f nrf52
Erasing user available code and UICR flash areas.
Applying system reset.

PS C:\FUN\NRF5\zephyrproject\zephyr> nrfjprog --program .\build\zephyr\zephyr.hex -f nrf52
Parsing hex file.
Reading flash area to program to guarantee it is erased.
Checking that the area to write is not protected.
Programming device.
`

Please advice how to use west to flash my device.
Thanks!

mbolivar, 2020-03-17:
Hello, thanks for your report. Sorry for the traceback; that is not very user friendly and I will try to fix it.

However, I think this is a problem with your board, not west flash.

> PS C:\FUN\NRF5\zephyrproject\zephyr>west build ....\ble_beacon_zephyr

In the future, please don't truncate command lines you typed.

Including as much detail as you can about what you did makes it easier to help you (see https://docs.zephyrproject.org/latest/guides/getting-help.html#how-to-ask)

> -- Board: nrf52832_mdk

Luckily, in this case, the complete cmake output gave me the information I needed, which was the board you are using.

This board only supports the pyocd runner:
https://github.com/zephyrproject-rtos/zephyr/blob/master/boards/arm/nrf52832_mdk/board.cmake

So west flash --runner jlink is not expected to work.

Please see the board porting guide for more information on adding runner support to a board.

closed 2020-03-17 by mbolivar

mbolivar, 2020-03-17:
> Sorry for the traceback; that is not very user friendly and I will try to fix it.

Hmm, what version of zephyr are you using? I can't reproduce this traceback.

On v2.2 and the latest master branch, I get:

```
$ west flash --runner jlink
-- west flash: rebuilding
ninja: no work to do.
-- west flash: using runner jlink
WARNING: runner jlink is not configured for use with nrf52832_mdk, this may not work
Traceback (most recent call last):
  File "/home/mbolivar/.virtualenvs/west-dev/bin/west", line 11, in <module>
    load_entry_point('west', 'console_scripts', 'west')()
  File "/home/mbolivar/src/west/src/west/app/main.py", line 768, in main
    app.run(argv or sys.argv[1:])
  File "/home/mbolivar/src/west/src/west/app/main.py", line 106, in run
    self.run_command(argv)
  File "/home/mbolivar/src/west/src/west/app/main.py", line 334, in run_command
    self.run_extension(args.command, argv)
  File "/home/mbolivar/src/west/src/west/app/main.py", line 397, in run_extension
    command.run(args, unknown, self.topdir, manifest=self.manifest)
  File "/home/mbolivar/src/west/src/west/commands.py", line 116, in run
    self.do_run(args, unknown)
  File "/home/mbolivar/zp/zephyr/scripts/west_commands/flash.py", line 30, in do_run
    do_run_common(self, my_args, runner_args)
  File "/home/mbolivar/zp/zephyr/scripts/west_commands/run_common.py", line 168, in do_run_common
    runner_config['args'][runner_name] +
KeyError: 'jlink'
```

In particular, I see this additional warning not present in the original report output:

```
WARNING: runner jlink is not configured for use with nrf52832_mdk, this may not work
```

Do you also get this warning? It explains the problem, so I think I'm just going to make it a fatal error instead of a warning.

Thanks again for the report and bringing this up; we should definitely make this output easier to read.

GeorgeGkinis, 2020-04-01:
I ended up defining my own board and altered the **board.cmake** file as follows :
```
#board_runner_args(nrfjprog "--nrf-family=NRF52")
board_runner_args(jlink "--device=nrf52" "--erase" "--speed=4000")
board_runner_args(pyocd "--target=nrf52" "--frequency=4000000")
#include(${ZEPHYR_BASE}/boards/common/nrfjprog.board.cmake)
include(${ZEPHYR_BASE}/boards/common/jlink.board.cmake)
include(${ZEPHYR_BASE}/boards/common/pyocd.board.cmake)
include(${ZEPHYR_BASE}/boards/common/openocd-nrf5.board.cmake)
```

Now **jlink** is the first to try and west flashes happily :)