zephyrproject-rtos/zephyr #23546: Kconfig: default value not assigned when inheriting Kconfig values in range

cvinayak, 2020-03-18:
**Describe the bug**
In the below snippet, CONFIG_BT_CTLR_ADV_AUX_SET is '' 

```
config BT_CTLR_ADV_SET
	int "LE Advertising Extensions Sets"
	depends on BT_CTLR_ADV_EXT
	range 1 64
	help
	  Maximum supported advertising sets.

config BT_CTLR_ADV_AUX_SET
	int "LE Advertising Extensions Auxiliary Sets"
	depends on BT_CTLR_ADV_EXT
	range 1 BT_CTLR_ADV_SET if BT_CTLR_ADV_PERIODIC
	range 0 BT_CTLR_ADV_SET
	help
	  Maximum supported advertising auxiliary channel sets.
```

**To Reproduce**

Use: https://github.com/zephyrproject-rtos/zephyr/pull/22237
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -GNinja -DBOARD=nrf51_pca10028 -DCONF_FILE=prj_controller_dbg_ll_sw_split.conf  ../tests/bluetooth/init 
3. ninja
4. See error output listed in following sections

**Expected behavior**
CONFIG_BT_CTLR_ADV_AUX_SET  shall be set to '0'.

**Impact**
annoyance and showstopper, needing explicit assignment.

**Screenshots or console output**
```
[26/192] Building C object zephyr/CMakeFiles/zephyr.dir/misc/generated/configs.c.obj
FAILED: zephyr/CMakeFiles/zephyr.dir/misc/generated/configs.c.obj 
ccache /home/jack/zephyr-sdk/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc -DBUILD_VERSION=zephyr-v2.2.0-465-g7c88e9915b44 -DKERNEL -DNRF51 -D_FORTIFY_SOURCE=2 -D__PROGRAM_START -D__ZEPHYR__=1 -I../../include -Izephyr/include/generated -I../../soc/arm/nordic_nrf/nrf51 -I../../ext/lib/crypto/tinycrypt/include -I../../ext/hal/cmsis/Core/Include -I../../subsys/testsuite/include -I../../subsys/testsuite/ztest/include -I../../subsys/bluetooth -I/home/jack/workspace/west_zephyr/modules/hal/nordic/nrfx -I/home/jack/workspace/west_zephyr/modules/hal/nordic/nrfx/drivers/include -I/home/jack/workspace/west_zephyr/modules/hal/nordic/nrfx/mdk -I/home/jack/workspace/west_zephyr/modules/hal/nordic/. -isystem ../../lib/libc/minimal/include -isystem /home/jack/zephyr-sdk/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/9.2.0/include -isystem /home/jack/zephyr-sdk/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/9.2.0/include-fixed -Og -imacros/home/jack/workspace/west_zephyr/zephyr/build/init/zephyr/include/generated/autoconf.h -ffreestanding -fno-common -g -mcpu=cortex-m0 -mthumb -imacros/home/jack/workspace/west_zephyr/zephyr/include/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wno-address-of-packed-member -Wno-pointer-sign -Wpointer-arith -Wno-unused-but-set-variable -Werror=implicit-int -fno-asynchronous-unwind-tables -fno-pie -fno-pic -fno-strict-overflow -fno-reorder-functions -fno-defer-pop -fmacro-prefix-map=/home/jack/workspace/west_zephyr/zephyr/tests/bluetooth/init=CMAKE_SOURCE_DIR -fmacro-prefix-map=/home/jack/workspace/west_zephyr/zephyr=ZEPHYR_BASE -ffunction-sections -fdata-sections -mabi=aapcs -march=armv6s-m -std=c99 -nostdinc -MD -MT zephyr/CMakeFiles/zephyr.dir/misc/generated/configs.c.obj -MF zephyr/CMakeFiles/zephyr.dir/misc/generated/configs.c.obj.d -o zephyr/CMakeFiles/zephyr.dir/misc/generated/configs.c.obj   -c zephyr/misc/generated/configs.c
In file included from ../../include/toolchain.h:39,
                 from zephyr/misc/generated/configs.c:7:
zephyr/misc/generated/configs.c: In function '_ConfigAbsSyms':
../../include/toolchain/gcc.h:399:51: error: expected expression before ')' token
  399 |   "\n\t.type\t" #name ",%%object" :  : "n"(~(value)))
      |                                                   ^
zephyr/misc/generated/configs.c:314:1: note: in expansion of macro 'GEN_ABSOLUTE_SYM'
  314 | GEN_ABSOLUTE_SYM(CONFIG_BT_CTLR_ADV_AUX_SET, );
      | ^~~~~~~~~~~~~~~~
[30/192] Building C object zephyr/CMakeFiles/zephyr.dir/soc/arm/nordic_nrf/validate_base_addresses.c.obj
ninja: build stopped: subcommand failed.
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK, 0.11.2
 - Commit SHA or Version used #22237

**Additional context**
Add any other context about the problem here.

cvinayak, 2020-03-19:
@ulfalizer any ETA? @SebastianBoe do you have any comments as to why a default value is not assigned from the range when the range uses another Kconfig value that gets default from its range?

SebastianBoe, 2020-03-19:
Is **BT_CTLR_ADV_EXT** true?

SebastianBoe, 2020-03-19:
What is the actual behaviour, that BT_CTLR_ADV_AUX_SET is unset?

It is expected behaviour that BT_CTLR_ADV_AUX_SET is unset when it's dependency BT_CTLR_ADV_EXT is not true.

cvinayak, 2020-03-19:
BT_CTLR_ADV_EXT is true and BT_CTLR_ADV_SET is assigned 1. But BT_CTLR_ADV_AUX_SET is unassigned.

SebastianBoe, 2020-03-19:
What about PERIODIC, is that true?

SebastianBoe, 2020-03-19:
Can you please upload the build_dir/zephyr/.config file ?

cvinayak, 2020-03-19:
There you have my console log and the config file (renamed to txt file). Notice `CONFIG_BT_CTLR_ADV_AUX_SET=`

[console_log.txt](https://github.com/zephyrproject-rtos/zephyr/files/4353666/console_log.txt)
[console_log_config.txt](https://github.com/zephyrproject-rtos/zephyr/files/4353667/console_log_config.txt)

SebastianBoe, 2020-03-19:
This is due to us not specifying a default value.

Adding "default 0" fixes the issue.

From the docs:

> Defaults should always be given for int and hex symbols, however, as they implicitly default to the empty string. This is partly for compatibility with the C Kconfig tools, though an implicit 0 default might be less likely to be what was intended compared to other symbol types as well.

https://docs.zephyrproject.org/1.14.0/guides/kconfig/index.html

Arguably, assigning the empty string to an int type should be treated as an error, but I'll leave that up to @ulfalizer .

cvinayak, 2020-03-19:
Whats is inconsistent is `BT_CTLR_ADV_SET` is assigned a default of 1 (based on the range option I assume), default has not been given explicitly in the Kconfig definition.

SebastianBoe, 2020-03-19:
I can't explain this, perhaps having no default value is undefined behaviour.

cvinayak, 2020-03-19:
I fix my PR with explicit default assignments (and conditionals).

I will let @ulfalizer to comment, and close the issue as he feel right.

closed 2020-07-17 by github-actions[bot]