zephyrproject-rtos/zephyr #23555: STM32 SDMMC disk access driver (based on stm32 cube HAL)

anthonybrandon, 2020-03-18:
I've written a disk access driver for the stm32 sdmmc component based around the stm32 cube HAL, and I am wondering if there is any interest in merging something like that.
The only related PR I could fine was #13214, but it seems to be abandoned.
If there is interest I will make a PR with my changes.

erwango, 2020-03-18:
Hi @anthonybrandon,

There is definitely an interest. Don't hesitate to base your work on https://github.com/zephyrproject-rtos/zephyr/pull/13214 which was of a good quality and had well split commits.

anthonybrandon, 2020-03-18:
@erwango #13214 introduces an API for the low level sdmmc driver (for which I am using stm32 cube). I can upload my change and maybe we can discuss how it should be changed.
If you prefer I push #13214 forward, that is also an option I guess.

anthonybrandon, 2020-03-18:
My PR: #23561

erwango, 2020-03-18:
Thanks. Let's discuss it there.

anthonybrandon, 2020-05-28:
@erwango Since my PR is merged I am interested in adding support for interrupts to the driver. Is it okay to discuss here? Or on a slack channel? My main question is if I should just replace the current polling implementation with an interrupt based one, or if it should be some Kconfig option. I'm not sure there is any reason to ever not use the interrupt based implementation if it was available.

erwango, 2020-05-28:
@anthonybrandon Let's discuss on Slack or you can also open a new ticket.
At first, I won't see any reason to keep the polling if interrupt is supported, but you can open a RFC and see if there are people concerned by this option.
This being said, this is a new driver so I don't think there will be much more users to complain.

I'm closing this issue as the work is now done.

closed 2020-05-28 by erwango