zephyrproject-rtos/zephyr #23577: [Coverity CID :208923] Out-of-bounds read in drivers/sensor/adxl345/adxl345.c

zephyrbot, 2020-03-18:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/master/drivers/sensor/adxl345/adxl345.c#L104

Category: Memory - illegal accesses
Function: `adxl345_channel_get`
Component: Drivers
CID: [208923](https://scan9.coverity.com/reports.htm#v29726/p12996/mergedDefectId=208923)

Details:

```
107         case SENSOR_CHAN_ACCEL_Z:
108             adxl345_accel_convert(val, data->bufz[data->sample_number]);
109             data->sample_number++;
110             break;
111         case SENSOR_CHAN_ACCEL_XYZ:
112             adxl345_accel_convert(val++, data->bufx[data->sample_number]);
>>>     CID 208923:    (OVERRUN)
>>>     Overrunning array "data->bufy" of 32 2-byte elements at element index 32 (byte offset 65) using index "data->sample_number" (which evaluates to 32).
113             adxl345_accel_convert(val++, data->bufy[data->sample_number]);
114             adxl345_accel_convert(val,   data->bufz[data->sample_number]);
115             data->sample_number++;
116             break;
117         default:
118             return -ENOTSUP;
98         switch (chan) {
99         case SENSOR_CHAN_ACCEL_X:
100             adxl345_accel_convert(val, data->bufx[data->sample_number]);
101             data->sample_number++;
102             break;
103         case SENSOR_CHAN_ACCEL_Y:
>>>     CID 208923:    (OVERRUN)
>>>     Overrunning array "data->bufy" of 32 2-byte elements at element index 32 (byte offset 65) using index "data->sample_number" (which evaluates to 32).
104             adxl345_accel_convert(val, data->bufy[data->sample_number]);
105             data->sample_number++;
106             break;
107         case SENSOR_CHAN_ACCEL_Z:
108             adxl345_accel_convert(val, data->bufz[data->sample_number]);
109             data->sample_number++;
```

Please fix or provide comments in coverity using the link:

https://scan9.coverity.com/reports.htm#v32951/p12996.

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

mateusz-holenko, 2020-03-20:
We'll look into this

carlescufi, 2020-04-22:
Fixed by #23931

closed 2020-04-22 by carlescufi