zephyrproject-rtos/zephyr #23588: [Coverity CID :208912] Dereference after null check in tests/net/icmpv4/src/main.c

zephyrbot, 2020-03-18:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/master/tests/net/icmpv4/src/main.c#L478

Category: Null pointer dereferences
Function: `test_icmpv4_send_echo_req_opt`
Component: Tests
CID: [208912](https://scan9.coverity.com/reports.htm#v29726/p12996/mergedDefectId=208912)

Details:

```
472    
473         pkt = prepare_echo_request_with_options(iface);
474         if (!pkt) {
475             zassert_true(false, "EchoRequest with opts packet prep failed");
476         }
477    
>>>     CID 208912:  Null pointer dereferences  (FORWARD_NULL)
>>>     Passing null pointer "pkt" to "net_ipv4_input", which dereferences it.
478         if (net_ipv4_input(pkt)) {
479             net_pkt_unref(pkt);
480             zassert_true(false, "Failed to send");
481         }
482     }
483    
```

Please fix or provide comments in coverity using the link:

https://scan9.coverity.com/reports.htm#v32951/p12996.

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

jukkar, 2020-03-18:
@rveerama1 can you check this one.

jukkar, 2020-03-19:
false positive

closed 2020-03-19 by jukkar