zephyrproject-rtos/zephyr #23599: zephyr/samples/application_development/code_relocation execution stop at z_arm_bus_fault

kkiyota, 2020-03-19:
When I ran code_relocation sample on my custom board, execution was stopped by vector catch fault at z_arm_bus_fault.  Debugger screen shot is shown below.  I'm not sure so far if this is related to board, soc, DTS, etc or hardware, 

Could you advise which one I should check? 

[image]

Thank you for your help!

kkiyota, 2020-03-19:
In addition, it could work on qemu_cortex_m3 and nucleo_f767zi, but not on my own board.  So I assume something wrong for my own board setting.  I need your suggestions which items I should check.

Thanks!

kkiyota, 2020-03-23:
I captured the following info when bus fault occurred.   Could you let me know how I can interpret it and how I can fix it?

[image]

carlescufi, 2020-05-21:
@wentongwu @ioannisg @andrewboie any ideas?

wentongwu, 2020-05-22:
@kkiyota please check the memory range for your board, sram2 section will be like this.
```
#define RAM_SIZE2 (CONFIG_SRAM_SIZE * 512)
MEMORY
    {
#ifdef CONFIG_SRAM2
    SRAM2                  (wx) : ORIGIN = (CONFIG_SRAM_BASE_ADDRESS + RAM_SIZE2), LENGTH = RAM_SIZE2
#endif
    }
```

andrewboie, 2020-07-22:
@ioannisg might have some insight here.

wentongwu, 2020-09-07:
@kkiyota ping

wentongwu, 2020-09-21:
@kkiyota ping

wentongwu, 2020-09-23:
@kkiyota any comment? Thanks

kkiyota, 2020-09-23:
@wentongwu Since time has passed, I don't have this setup anymore.  So I need to decline it.

Thank you for your follow up!

wentongwu, 2020-09-24:
@kkiyota ok, I will close this issue, please reopen it if any new finding, Thanks.

closed 2020-09-24 by wentongwu