zephyrproject-rtos/zephyr #23611: Add QuickLogic EOS S3 HAL west module

PiotrZierhoffer, 2020-03-19:
Antmicro is working jointly with QuickLogic on adding their platform support to upstream Zephyr.

While working on https://github.com/zephyrproject-rtos/zephyr/pull/23565 it was pointed out that the HAL module is supposed to be hosted by zephyrproject-rtos.

I would like to propose adding a repository for this module. It currently resides here: https://github.com/antmicro/eos-s3-hal

It's licensed on Apache 2.0 and originates from QuickLogic.

I suggest we use the name `hal_quicklogic` for the repo.

More info on the platform here: https://www.quicklogic.com/products/eos-s3/

cc @carlescufi

carlescufi, 2020-03-19:
@galak @MaureenHelm @nashif @ioannisg @andrewboie @jhedberg any objections?

+1 from me on the name: `hal_quicklogic`

andrewboie, 2020-03-19:
Fine w/me.

carlescufi, 2020-03-25:
@PiotrZierhoffer before I create the repo, just a formality: you are committing to maintain this platform for the foreseeable future in the upstream Zephy repos, correct?

mgielda, 2020-03-27:
Hi @carlescufi we communicated back with the manufacturer (QuickLogic) as this is work performed towards them. They acknowledge they will maintain the port with us and keep it compatible with the Zephyr code base as it evolves.

PiotrZierhoffer, 2020-04-01:
@carlescufi if that confirmation is enough, we'd love to move this forward, as I'll be cleaning up the PR today as well. Can you please help with the repo?

carlescufi, 2020-04-09:
Sorry for the delay.

Created the repo and added @PiotrZierhoffer to it:
https://github.com/zephyrproject-rtos/hal_quicklogic

closed 2020-04-09 by carlescufi