zephyrproject-rtos/zephyr #23618: cmake: Export compile_commands.json for all generated code

sslupsky, 2020-03-19:
Presently, if you include the following setting in a CMakeLists.txt file in the project root, the targets listed in the compile_commands.json file generated by CMake are limited to the targets specified in the CMakeLists.txt file.
```
set(CMAKE_EXPORT_COMPILE_COMMANDS ON)
```
I would prefer that the compile_commands.json file also include all the compiled code for Zephyr (the drivers, kernel, subsys, etc).  There reason is that I can configure MS VSCode to read the compile_commands.json file and configure its intellisense feature to highlight coding errors.

I tried editing the CMakeLists.txt files in the following Zephyr tree subfolders:
arch/arm
boards
drivers
kernel
soc
subsys

This improved things quite a bit and many of the Zephyr targets now appear in the compile_commands.json file.  However, there are a few that still do not appear.  For instance, sam0_rtc_timer.c is not included as a target.  Neither is wdt_sam0.c.  I know these are being compiled because my application uses them and I have debugged them.  There are likely a few other files that are not included as well, I have not yet performed an exhaustive analysis.

Is there another way to inform CMake so that a complete set of targets will be included in compile_commands.json?

mbolivar-nordic, 2020-03-21:
> Is there another way to inform CMake so that a complete set of targets will be included in compile_commands.json?

There are several examples in the west documentation that I believe do what you are asking about:

https://docs.zephyrproject.org/latest/guides/west/build-flash-debug.html#permanent-cmake-arguments

sslupsky, 2020-03-22:
Thank you @mbolivar 

I was able to get export setting to stick by adding
```
-DCMAKE_EXPORT_COMPILE_COMMANDS=1
```
to the west command line.

There is still a VSCode CMake-Tools issue where resolving the location of a header file can result is multiple paths.  I have noticed this happens when there are headers with no target.  That is, there is a <kernel.h> but there is not target "kernel.c".  CMake-Tools attempts to look up "kernel.c" when I try to local <kernel.h>  Since there is nothing to look up in the compile_commands.json, it defaults to some other search process and comes up with multiple paths.

closed 2020-03-23 by SebastianBoe