zephyrproject-rtos/zephyr #23622: litex_vexriscv: k_busy_wait() never returns if called with interrupts locked

stephanosio, 2020-03-20:
**Describe the bug**
k_busy_wait() never returns if called with interrupts locked on the `litex_vexriscv` platform, which uses `litex_timer` system timer.

For more details, refer to the issue #23519.

stephanosio, 2020-03-20:
cc @mateusz-holenko

carlescufi, 2020-04-07:
@mateusz-holenko do you have plans to address this?

mateusz-holenko, 2020-04-07:
@carlescufi  - yes, we are planning to look at this soon.

stephanosio, 2020-05-11:
Has this been addressed?

mateusz-holenko, 2020-05-12:
We are currrently working on this. It turned out that the timer in LiteX is very simple and implementing a proper `k_busy_wait()` (allowing to wait for periods longer than 10ms) would require either a custom not-trivial implementation or extending the capabilities of HW (since it's a soft IP core anyway).

We decided to go with the second option and are now adding a register allowing to read the total number of timer ticks from reset to the timer itself. Once it's done and tested we will update the driver to use this in `k_cycle_get32()` function.

PR #25418 (2020-05-26): Fix locked IRQ support in LiteX timer
This reimplements z_timer_cycle_get_32() so it works
when IRQs are locked and solves the hung
k_busy_wait() problem.

This fixes #23622.
commit 265d2cfa (2020-05-26): drivers: litex_timer: Fix IRQ lock intolerance

This reimplements z_timer_cycle_get_32() so it works
when IRQs are locked and solves the hung
k_busy_wait() problem.

Fixes #23622.
closed 2020-05-26 by carlescufi