zephyrproject-rtos/zephyr #23629: support inverted PWM on STM32

javier-moreno-tridonic-com, 2020-03-20:
Facing the problem where I need to build the same codebase for a nordic uC and an stm32f4 uC.

Even though this was merged in recently

https://github.com/zephyrproject-rtos/zephyr/pull/20657/files#diff-722b2739b6d0b69a16ee867da52074d6R58

Looks like for stm32 targets, pwm flag usage is not supported

https://github.com/vestas-wind-systems/zephyr/blob/162a49d963f0b9b94c3e25ef8de75f89f7fb4a71/drivers/pwm/pwm_stm32.c#L87-L90

One could #ifdef the application code to provide 1-period as value for the PWM signal. But Since the mechanism is there, would be nice to harmonize usage across targets.

@erwango 
@rsalveti 
@idlethread
Is there any plans to support this for stm?

erwango, 2020-03-20:
@javier-moreno-tridonic-com There is no plan to update PWM driver to support flags yet, but any contribution to add it would be welcome!

erwango, 2020-05-25:
@javier-moreno-tridonic-com I'm closing the point. Please reopen if need be

gmarull, 2020-05-25:
@erwango are there any plans for merging #24569 after 2.3.0 release? If so, I'd leave this open until the functionality gets merged.

erwango, 2020-05-25:
> @erwango are there any plans for merging #24569 after 2.3.0 release? If so, I'd leave this open until the functionality gets merged.

@gmarull Sure!

PR #24569 (2020-06-19): Refactor STM32 PWM driver using LL API
The PWM driver has been refactored using the HAL LL API. Not only that, but the set pin_set function is now faster, as channel output compare is just initialized if needed.

Support for the polarity flag has also been added to the STM32 PWM driver. STM32 boards using PWM have been updated accordingly.

Fixes #23629
closed 2020-06-19 by carlescufi