zephyrproject-rtos/zephyr #23643: drivers: kscan: ft5336: add multitouch gestures etc.

k0d, 2020-03-20:
**Is your enhancement proposal related to a problem? Please describe.**
It appears that there are some gesture detection features in the ft5336 IC. These should be implemented.

**Describe the solution you'd like**
Not sure yet, this needs more research.

Registers are listed here.
https://github.com/STMicroelectronics/STM32CubeF7/blob/master/Drivers/BSP/Components/ft5336/ft5336.h
This could be useful also.
https://www.newhavendisplay.com/datasheets.html (application_notes has registers)

k0d, 2020-03-20:
@MaureenHelm I intend to implement more missing features from this. Just thought you'd like to know.

gmarull, 2023-05-30:
Obsolete, this can now be handled using the input API.

closed 2023-05-30 by gmarull

JohnAZoidberg, 2026-01-03:
See:

- https://github.com/zephyrproject-rtos/zephyr/pull/70631
- https://github.com/petejohanson/maxtouch-zephyr-module
- https://github.com/zephyrproject-rtos/zephyr/commit/ad254b06c8343fb607e6ab352b40c7c20dc4eff7