zephyrproject-rtos/zephyr #23662: Building blinky sample program goes wrong

nonopx, 2020-03-21:
I have followed https://docs.zephyrproject.org/2.2.0/getting_started/index.html guide in order to
download zephyr.

On step 6.2 - west build -p auto -b <your-board-name> samples/basic/blinky
(I put esp32 although I do not have the hardware yet. just trying to build the program)

I was excepting the build to go right but instead, I'm getting this error message (screenshot ):

[image]
[image]

Environment:
 - OS: (Linux VM on windows host)
 - Toolchain (Zephyr SDK)
 - Zephyr version: 2.2.99

nashif, 2020-03-23:
Zephyr SDK does not support building the ESP32, for that, you need to download the toolchain from espressif and setup the toolchain according to the board documentation.

jenmwms, 2020-03-23:
@nonopx Here is some documentation with that info for the ESP32 board in Zephyr - https://docs.zephyrproject.org/latest/boards/xtensa/esp32/doc/index.html
Check out the https://docs.zephyrproject.org/latest/boards/xtensa/esp32/doc/index.html#system-requirements

nonopx, 2020-03-23:
Thank you very much

closed 2020-03-23 by nonopx