zephyrproject-rtos/zephyr #23669: question: is zephyr supported ipv4 rx fragments

ehudmarvell, 2020-03-22:
Hi,

I am running the latest OS zephyr and I have issue on ipv4 while I have rx fragments.
So I try to understand from the docs\code if it supports but didn't find any reference for enable\disable it. I did see a reference for Ipv6 fragments.
Is Ipv4 rx fragments reassembly is supported? If yes, can you add details?

Thanks,
Ehud

jukkar, 2020-03-23:
IPv4 packet fragmentation is not supported.

ehudmarvell, 2020-03-24:
Thank you @jukkar , do you know if it is plan on road map?

jukkar, 2020-03-24:
There has not been any discussion whether we should have that feature or not. This is first time someone asking this. You can always submit a feature request issue for any missing feature, but if no one else needs that feature, you should be prepared to submit patches yourself.

closed 2020-03-24 by ehudmarvell