zephyrproject-rtos/zephyr #23672: dts: samd: question: Is it possible to clean up samd.dtsi devicetree warning?

sslupsky, 2020-03-22:
When building a project using the samd dts, an warning is generated because of a duplicate unit address in dts/arm/atmel/samd.dtsi.  The warning can be safely ignored because the samd PORT peripheral controls both the GPIO and the PINMUX.  I am wondering if the warning can be cleanup up somehow?

Here is the warning
```
arduino_mkr_1310.dts.pre.tmp:87.29-91.5: Warning (unique_unit_address_if_enabled): /soc/pinmux@41004400: duplicate unit-address (also used in node /soc/gpio@41004400)
arduino_mkr_1310.dts.pre.tmp:92.29-96.5: Warning (unique_unit_address_if_enabled): /soc/pinmux@41004480: duplicate unit-address (also used in node /soc/gpio@41004480)
```

carlescufi, 2020-05-21:
@galak could you take a stab at answering the question please?

stephanosio, 2020-05-21:
Atmel SAM0 GPIO, in general, requires a major overhaul, and this problem should probably be fixed as part of that (at this time, pinmux-ing is not properly done in the DT, but in a hack file `pinmux.c`, and this is undesirable).

nzmichaelh, 2020-07-15:
@stephanosio out of interest, what does a proper DT based pinmux setup look like?

sslupsky, 2020-07-18:
I attempted to suppress the warning by adding the following line to pre_dt_board.cmake:
```
list(APPEND EXTRA_DTC_FLAGS "-Wno-unique_unit_address")
```
but this generates a compiler error:
```
gen_defines.py: error: unrecognized arguments: -Wno-unique_unit_address'
```
Similarly, 
```
list(APPEND EXTRA_DTC_FLAGS "-Wno-unique_unit_address_if_enabled")
```
generates error:
```
gen_defines.py: error: unrecognized arguments: -Wno-unique_unit_address_if_enabled'
```

stephanosio, 2020-07-20:
> @stephanosio out of interest, what does a proper DT based pinmux setup look like?

@nzmichaelh Something like:
https://github.com/zephyrproject-rtos/zephyr/blob/98bcc51b0925a92db28f127d08b842b1d609bce8/dts/arm/atmel/same70.dtsi#L241-L301

https://github.com/zephyrproject-rtos/zephyr/blob/98bcc51b0925a92db28f127d08b842b1d609bce8/dts/arm/atmel/same70.dtsi#L216

nzmichaelh, 2020-07-20:
@stephanosio thanks.  I'm not sure if I'll finish it, but I started a branch at https://github.com/zephyrproject-rtos/zephyr/compare/master...nzmichaelh:pinctrl

closed 2020-09-08 by sslupsky