zephyrproject-rtos/zephyr #23674: Openthread stop working after "Update OpenThread revision #23632"

retfie, 2020-03-22:
**Describe the bug**
Build lwm2m_client sample with OpenThread support, and manually commission it via shell.
Instead of joining already available network, it creates a new segment with the same network parameters and set his state as Leader instead of Child as with before this change. Besides it gets IP from the same local-mesh segment It's unable to ping border router IP or any node in this network.

**To Reproduce**
cp samples/net/sockets/echo_client/overlay-ot.conf samples/net/lwm2m_client
west build -b nrf52840_pca10056  samples/net/lwm2m_client -- -DCONF_FILE="prj.conf overlay-ot.conf"
west flash --hex-file build/zephyr/zephyr.hex

ot thread stop
ot channel <channel>
ot paind <panid>
ot extpanid <extpanid>
ot masterkey <masterkey>
ot networkname <networkname>
ot thread start

Look for messages like:
<inf> net_l2_openthread: State changed! Flags: 0x000012a4 Current role: 2

**Expected behavior**
Current role, should change from 0 up to 2 or 3 and join existing network (in very close proximity) and not going to 4 (Leader) and form a new one, not reachable from other nodes in this network.

When module openthread is checkout at it's previous hash (3c6191eb4e8ca44b5f4eeaa696837bce14e83c69) everything works fine and node was able to connect and register with LWM2M server.

**Impact**
Openthread is not functional with this change.

**Environment (please complete the following information):**
 - OS: Linux
 - Zephyr version: 2.2.99
 - Commit SHA: 0095daefbc5b1f73c531e1b1e3e615c6b1647ee0

rlubos, 2020-03-23:
Hi, there was a change in the usptream OpenThread in default mbedTLS configuration, this can cause compability issues (in general cryptography does not work as expected). I'll open a PR with a fix soon.

rlubos, 2020-03-23:
Please give https://github.com/zephyrproject-rtos/zephyr/pull/23688 a try.

retfie, 2020-03-23:
Hi Robert,
this PR #23688  fixes the problem, thanks !

closed 2020-03-23 by retfie

lualven-duplicate, 2020-03-25:
@retfie Hi, did you get the on-mesh prefixed address (fd11:22) from BR? I cannot on my side and do not know why. #23703

retfie, 2020-03-28:
@lualven-duplicate sorry for late response, but yes, I've got on-mesh prefixed address, and I can ping it from BR.