zephyrproject-rtos/zephyr #23686: kernel: API: suggestion to remove k_cpu_atomic_idle

ioannisg, 2020-03-23:
## Introduction

`k_cpu_atomic_idle()` API is not used anywhere in Zephyr, contrary to `k_cpu_idle()` which is used by the idle thread and, possibly, in device power management. This is a suggestion to remove it.

### Problem description

Essentially, the differences between k_cpu_atomic_idle() and k_cpu_idle() are two.
- k_cpu_idle() un-conditionally un-masks interrupts upon return, while k_cpu_atomic_idle(key) gets the interrupt locking state as argument and restores the value upon return. This could be, anyways, moved to the called of the API function - we might not need to have 2 kernel APIs for that.

- k_cpu_atomic_idle() is "specified" as "to be called with interrupts locked". However, in the idle thread, k_cpu_idle() is _anyways_ called with interrupts locked, so the same behavior is achieved without using the _atomic_ API.

### Proposed change

Suggest to remove k_cpu_atomic_idle() API.

The fact that the API is not used in Zephyr sub-systems / drivers makes it easy to depreciate it now.

ioannisg, 2020-03-23:
CC @pabigot @andyross @andrewboie @stephanosio @wentongwu

andrewboie, 2020-03-24:
+1 from me

stephanosio, 2020-03-24:
Agreed.

As long as `arch_cpu_idle` requires interrupts to be disabled/locked before it is called (and this is a de-facto requirement given that the power management subsystem requires `_kernel.idle` to be set before calling this function to ensure proper operation, and ISR checks `_kernel.idle` and calls `z_sys_power_save_idle_exit` when set), there is no real value in having `arch_cpu_atomic_idle`.

pabigot, 2020-03-24:
> The fact that the API is not used in Zephyr sub-systems / drivers makes it easy to depreciate it now.

This doesn't mean nobody's depending on it.  We have a whole device power management API with no in-tree use that is being used in proprietary out of tree applications.

That aside, I have no objection to removing this function.  The current situation could use certainly some reduction in confusion.

ioannisg, 2020-03-25:
Side note: If we prefer to keep k_cpu_atomic_idle() we will need to specify the API in a better way - right now its specification/implementation does not have fundamental differences from k_cpu_idle()

I'd like to here opinions from @andyross and @wentongwu as well :)