zephyrproject-rtos/zephyr #23694: TEMP_KINETIS is forced enabled on frdm_k64f if SENSORS is enabled. But ADC is missing

KimBoendergaard, 2020-03-23:
**Describe the bug**
When ever SENSORS is activate in a project based on frdm_k64f TEMP_KINETIS is auto selected.
The file temp_kinetisc thus being compiled in throws an err message if ADC (or more precise ADC_1) is not enabled.

**To Reproduce**
Use sample hello_world.
In menuconfig enable sensors and enable log
Build and see
 `<err> temp_kinetis: could not get ADC device` 
on console

If ADC is enabled in menuconfig, the error disappear due to "ADC_1" being defined

**Expected behavior**
In general I don't think features not explicitly wanted by the user should go into the system.
If it is acceptable, at least it should work without generating noise

**Impact**
Just annoyance and time wasted on tracing the root cause for the error message
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK
 - Commit SHA or Version used 9518bd19b7 (Release 2.2.0)

henrikbrixandersen, 2020-03-23:
You have altered the default configuration of the board to get into this situation. CONFIG_ADC_1 is enabled by default for this board.

KimBoendergaard, 2020-03-24:
It is correct ADC_1 is enabled by default, but it depends on CONFIG_ADC (in board-defconfig) which is not enabled by default and does not become it neither by enabling CONFIG_SENSORS

Sendt fra min Huawei Mobil

-------- Oprindelig meddelelse --------
Emne: Re: [zephyrproject-rtos/zephyr] TEMP_KINETIS is forced enabled on frdm_k64f if SENSORS is enabled. But ADC is missing (#23694)
Fra: Henrik Brix Andersen
Til: zephyrproject-rtos/zephyr
CC: Kim Bøndergaard ,Author

You have altered the default configuration of the board to get into this situation. CONFIG_ADC_1 is enabled by default for this board.

henrikbrixandersen, 2020-03-24:
> It is correct ADC_1 is enabled by default, but it depends on CONFIG_ADC (in board-defconfig) which is not enabled by default and does not become it neither by enabling CONFIG_SENSORS 

`CONFIG_TEMP_KINETIS` also depends on `CONFIG_ADC`. Anyways, I have already submitted a PR with a fix for the default setting of `CONFIG_TEMP_KINETIS` when `CONFIG_ADC` is enabled but `CONFIG_ADC_1` is not.

KimBoendergaard, 2020-03-24:
I partly agree. In menuconfig I see this for CONFIG_TEMP_KINETIS:

Direct dependencies (=y):
     SENSOR(=y) && BOARD_FRDM_K64F(=y)  (=y)
  || ADC(=n) && SOC_FAMILY_KINETIS(=y) && SENSOR(=y)  (=n)

I.e. I get CONFIG_TEMP_KINETIS by just choosing SENSOR.

But great you have a fix in the pipe line

Kim Bøndergaard
Prevas A/S
Team Manager / Systems Architect

Hedeager 3, DK-8200 Aarhus N
Phone +45 3315 9090
Mobile +45 5154 3961
kibo@prevas.dk
www.prevas.dk<https://webmail.prevas.se/owa/redir.aspx?C=iN_3zIhizUqp7BVo33RNR3R2ecTf_dEIdGI1j2JStIzDh7zuxYMXNtw7TKp3qanuPKuw20ZD0vE.&URL=http%3a%2f%2fwww.prevas.dk%2f>

PR #23696 (2020-03-24): NXP: Kinetis: Only enable temperature sensor driver by default if corresponding ADC is enabled
Only enable the NXP Kinetis temperature sensor driver by default if corresponding ADC is enabled. Fixes #23694.
commit 7427120e (2020-03-24): boards: arm: frdm_k64f: default enable for temp driver depends on adc

Only enable the NXP Kinetis temperature driver by default if ADC_1 is
enabled.

Fixes #23694.
closed 2020-03-24 by MaureenHelm