zephyrproject-rtos/zephyr #23702: STACK_POINTER_RANDOM is not working on ARM for the main thread

ceolin, 2020-03-24:
On arm (and possible others arch with custom swap to main) the main stack is not randomized, because `arch_switch_to_main_thread()` is called with `K_THREAD_STACK_SIZEOF(z_main_stack)` in switch_to_main_thread(), completely ignoring the value that was previous randomized in `z_setup_new_thread()`. The problem is that this value is not available anymore, except if it is build with `CONFIG_THREAD_STACK_INFO`

ceolin, 2020-03-24:
Possible solutions:

1) Make `STACK_POINTER_RANDOM` depends on `CONFIG_THREAD_STACK_INFO` for this platform.
2) change `z_setup_new_thread` to return this info
3) randomize stack before call `zetup_new_thread` -> the worst probably
4) Add stack size to  `k_thread` independently of `CONFIG_THREAD_STACK_INFO`

Other possibilities ? Thoughts ?

ceolin, 2020-03-24:
@andrewboie @andyross @ioannisg

ioannisg, 2020-03-25:
(1) and (2) seem OK to me, (4) as well, if we don't care about the extra bytes...
Didn't take a detailed look yet, though

stephanosio, 2020-05-22:
I haven't followed this up closely; but at first glance, this seems to be being addressed in #24714, is that correct? @ceolin @andrewboie

andrewboie, 2020-05-24:
@stephanosio yes correct, I incorporated @ceolin's idea into my series.

My large stack overhaul PR is not landing until 2.4, so if we strongly desire a fix for 2.3 just use @ceolin's PR that he closed and I'll rebase on it later

ioannisg, 2020-05-26:
> @stephanosio yes correct, I incorporated @ceolin's idea into my series.
> 
> My large stack overhaul PR is not landing until 2.4, so if we strongly desire a fix for 2.3 just use @ceolin's PR that he closed and I'll rebase on it later

That's fine with me.

ioannisg, 2020-07-31:
@andrewboie pls, clarify if this issue can be consider closed after we merged #24714

andrewboie, 2020-07-31:
yeah this shouldn't be a problem anymore.

closed 2020-07-31 by andrewboie