zephyrproject-rtos/zephyr #23703: Openthread on Zephyr cannot get On-Mesh Prefix address

lualven-duplicate, 2020-03-24:
**Describe the bug**
The openthread on zephyr build ok, all functions seems right , but it can not get the address by On-Mesh Prefix, unlike the original openthread build.

**To Reproduce**
Steps to reproduce the behavior:
1. cd echo_server
2. west build -b nrf52840_pca10056 -d build_nrf52840_pca10056_ot -- -DCONF_FILE="prj.conf overlay-ot.conf"
3. west flash -d build_nrf52840_pca10056_ot
4. from nrf52840_pca10056 cli: ot ipaddr
4. See error

**Expected behavior**
The ipaddr output should include fd11:22: address; which assigned by border router.

Below is correct output from original Openthread build (pca10056 #1):

```
> ipaddr
fd11:22:0:0:e062:9866:d140:eff2
fdde:ad00:beef:0:0:ff:fe00:2c00
fdde:ad00:beef:0:c67:3ba5:d158:2a99
fe80:0:0:0:a5:46ab:fed0:802f
Done
```

Below is the output from zephyr build (pca10056 #2):

```
uart:~$ ot ipaddr
fdde:ad00:beef:0:0:ff:fe00:1400
fdde:ad00:beef:0:0:0:0:2
fdde:ad00:beef:0:1a66:f554:9ac6:4a35
fe80:0:0:0:808a:c8fb:7566:979d
Done
uart:~$ net ipv6
IPv6 support                              : enabled
IPv6 fragmentation support                : disabled
Multicast Listener Discovery support      : enabled
Neighbor cache support                    : enabled
Neighbor discovery support                : enabled
Duplicate address detection (DAD) support : enabled
Router advertisement RDNSS option support : enabled
6lo header compression support            : disabled
Max number of IPv6 network interfaces in the system          : 1
Max number of unicast IPv6 addresses per network interface   : 8
Max number of multicast IPv6 addresses per network interface : 8
Max number of IPv6 prefixes per network interface            : 2

IPv6 addresses for interface 0x20016e80 (OpenThread)
====================================================
Type            State           Lifetime (sec)  Address
autoconf        preferred       infinite        fe80::808a:c8fb:7566:979d/128
autoconf        preferred       infinite        fdde:ad00:beef:0:1a66:f554:9ac6:4a35/128
autoconf        preferred       infinite        fe80::31ca:109e:1f3d:3492/128
manual          preferred       infinite        fdde:ad00:beef::2/128
uart:~$ net ipv6
IPv6 support                              : enabled
IPv6 fragmentation support                : disabled
Multicast Listener Discovery support      : enabled
Neighbor cache support                    : enabled
Neighbor discovery support                : enabled
Duplicate address detection (DAD) support : enabled
Router advertisement RDNSS option support : enabled
6lo header compression support            : disabled
Max number of IPv6 network interfaces in the system          : 1
Max number of unicast IPv6 addresses per network interface   : 8
Max number of multicast IPv6 addresses per network interface : 8
Max number of IPv6 prefixes per network interface            : 2
```

**Impact**
showstop, I can not access external IP address.
I tried several versions, all have such issue.

**Screenshots or console output**

**Environment (please complete the following information):**
 - OS: MacOS
 - Toolchain: Zephyr SDK
 - Commit SHA or Version used: 
v2.2-branch, d39cb42d0920d5658fad358ad5b91de75d747a20
v2.1-branch, b413223a66ed9e857c67ee21d1e8be4768804dd1
master, 669cf582e95cf48d6f6f90cdc5ded76b63e270fc

**Additional context**
Add any other context about the problem here.

rlubos, 2020-03-24:
@lualven-duplicate There is a PR pending for OT, fixing broken cryptography (which got messed up with the recent OT update). You could give it a go: https://github.com/zephyrproject-rtos/zephyr/pull/23688

It may seem unrelated, but as far as I can see, your Zephyr device became a leader and formed it's own mesh instead of joining the existing one (that's a result of broken crypto). That way, it could not obtain global prefix from BR, as they're in different meshes.

lualven-duplicate, 2020-03-25:
I do check the node, it can join the BR successfully, and first become a child, then router; and I can ping the BR's wpan0 address. So I don't think it is a different network. 
Any thoughts for any other problem?

lualven-duplicate, 2020-03-25:
This is the out put from node, it is  can ping wpan0, but not "fd11:22".
```
*** Booting Zephyr OS build zephyr-v2.1.0  ***
[00:00:00.000,000] <inf> ieee802154_nrf5: nRF5 802154 radio initialized
[00:00:00.006,835] <inf> net_l2_openthread: State changed! Flags: 0x011fd319 Current role: 0
[00:00:00.007,141] <inf> net_l2_openthread: OpenThread version: OPENTHREAD/g882e7074; NONE; Mar 25 2020 10:49:17
[00:00:00.007,141] <inf> net_l2_openthread: Network name: OpenThread
[00:00:00.007,171] <inf> net_l2_openthread: State changed! Flags: 0x00000004 Current role: 1
[00:00:00.007,232] <inf> net_l2_openthread: State changed! Flags: 0x00000001 Current role: 1
[00:00:00.007,354] <inf> net_l2_openthread: State changed! Flags: 0x00001000 Current role: 1
[00:00:00.007,476] <inf> net_l2_openthread: State changed! Flags: 0x00001000 Current role: 1
[00:00:00.007,598] <inf> net_l2_openthread: State changed! Flags: 0x00000010 Current role: 1
[00:00:00.007,629] <inf> net_l2_openthread: State changed! Flags: 0x00000020 Current role: 1
[00:00:00.013,641] <inf> net_config: Initializing network
[00:00:00.013,793] <inf> net_l2_openthread: State changed! Flags: 0x00000001 Current role: 1
[00:00:00.561,096] <inf> net_l2_openthread: State changed! Flags: 0x000010e4 Current role: 3
[00:00:00.574,218] <inf> net_l2_openthread: State changed! Flags: 0x00000200 Current role: 3
[00:00:03.013,977] <inf> net_echo_server_sample: Run echo server
[00:00:03.014,007] <inf> net_echo_server_sample: Network connected
[00:00:03.014,068] <inf> net_echo_server_sample: Starting...
[00:00:03.014,129] <inf> net_echo_server_sample: Waiting for UDP packets on port 4242 (IPv6)...
uart:~$ ot version
OPENTHREAD/g882e7074; NONE; Mar 25 2020 10:49:17
Done
uart:~$ 
uart:~$ ot state
router
Done
uart:~$ ot ipaddr
fdde:ad00:beef:0:0:ff:fe00:c00
fdde:ad00:beef:0:0:0:0:2
fdde:ad00:beef:0:4de9:3c0b:ccd1:6ff2
fe80:0:0:0:40f3:6114:3bac:6581
Done
uart:~$ ot ipaddr
fdde:ad00:beef:0:0:ff:fe00:c00
fdde:ad00:beef:0:0:0:0:2
fdde:ad00:beef:0:4de9:3c0b:ccd1:6ff2
fe80:0:0:0:40f3:6114:3bac:6581
Done
uart:~$ ot ping fd11:22::1639:6fb3:7736:feff
uart:~$ ot ping fdde:ad00:beef:0:df8f:25cc:fd27:d3c7
16 bytes from fdde:ad00:beef:0:df8f:25cc:fd27:d3c7: icmp_seq=2 hlim=64 time=36ms

```

This is my wpan0 info on BR:
```
wpan0: flags=4305<UP,POINTOPOINT,RUNNING,NOARP,MULTICAST>  mtu 1280
        inet6 fe80::5c6:4ae6:6372:98b6  prefixlen 64  scopeid 0x20<link>
        inet6 fe80::2c6c:35c5:155a:e11a  prefixlen 64  scopeid 0x20<link>
        inet6 fd11:22::1639:6fb3:7736:feff  prefixlen 64  scopeid 0x0<global>
        inet6 fdde:ad00:beef:0:df8f:25cc:fd27:d3c7  prefixlen 64  scopeid 0x0<global>
        unspec 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  txqueuelen 500  (UNSPEC)
        RX packets 2  bytes 128 (128.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 82  bytes 5187 (5.0 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

```

lualven-duplicate, 2020-03-25:
@rlubos Compared the compile flag from original OpenThread, I found the root cause: need to define the OPENTHREAD_CONFIG_IP6_SLAAC_ENABLE=1.

However, current zephyr cannot define it anywhere (CONFIG_OPENTHREAD_SLAAC is not defined in KConfig), I manual changed in modules/lib/openthread/zephyr/CMakeLists.txt:
set(OT_SLAAC ON CACHE BOOL "Enable SLAAC")

After this change, ping external ip function is ok, hope the fix can added to master branch ASAP.
```
uart:~$ ot ipaddr
fd11:22:0:0:32d1:2346:9a15:a293
fdde:ad00:beef:0:0:ff:fe00:e400
fdde:ad00:beef:0:0:0:0:2
fdde:ad00:beef:0:f09:d0dd:58d0:4d5e
fe80:0:0:0:1c29:6b17:2479:c03c
Done
uart:~$ net iface

Interface 0x20015880 (OpenThread) [1]
=====================================
Link addr : 8D:36:75:A5:D4:71:53:AE
MTU       : 1280
IPv6 unicast addresses (max 6):
        fe80::1c29:6b17:2479:c03c autoconf preferred infinite
        fdde:ad00:beef:0:f09:d0dd:58d0:4d5e autoconf preferred infinite meshlocal
        fdde:ad00:beef::2 manual preferred infinite meshlocal
        fd11:22::32d1:2346:9a15:a293 autoconf preferred infinite
IPv6 multicast addresses (max 8):
        ff02::1
        ff03::1
        ff03::fc
        ff32:40:fdde:ad00:beef::1
        ff33:40:fdde:ad00:beef::1
        ff02::2
        ff03::2
IPv6 prefixes (max 2):
        <none>
IPv6 hop limit           : 64
IPv6 base reachable time : 30000
IPv6 reachable time      : 35661
IPv6 retransmit timer    : 0
uart:~$ net ipv6
IPv6 support                              : enabled
IPv6 fragmentation support                : disabled
Multicast Listener Discovery support      : disabled
Neighbor cache support                    : disabled
Neighbor discovery support                : disabled
Duplicate address detection (DAD) support : disabled
Router advertisement RDNSS option support : disabled
6lo header compression support            : disabled
Max number of IPv6 network interfaces in the system          : 1
Max number of unicast IPv6 addresses per network interface   : 6
Max number of multicast IPv6 addresses per network interface : 8
Max number of IPv6 prefixes per network interface            : 2

IPv6 addresses for interface 0x20015880 (OpenThread)
====================================================
Type            State           Lifetime (sec)  Address
autoconf        preferred       infinite        fe80::1c29:6b17:2479:c03c/128
autoconf        preferred       infinite        fdde:ad00:beef:0:f09:d0dd:58d0:4d5e/128
manual          preferred       infinite        fdde:ad00:beef::2/128
autoconf        preferred       infinite        fd11:22::32d1:2346:9a15:a293/128

uart:~$ net ping 2001:db8:1:ffff::8.8.8.8
PING 2001:db8:1:ffff::8.8.8.8
8 bytes from 2001:db8:1:ffff::808:808 to fd11:22::32d1:2346:9a15:a293: icmp_seq=0 ttl=43 rssi=0 time=79 ms
8 bytes from 2001:db8:1:ffff::808:808 to fd11:22::32d1:2346:9a15:a293: icmp_seq=1 ttl=43 rssi=0 time=78 ms
8 bytes from 2001:db8:1:ffff::808:808 to fd11:22::32d1:2346:9a15:a293: icmp_seq=2 ttl=43 rssi=0 time=78 ms
uart:~$ 
```

rlubos, 2020-03-25:
@lualven-duplicate There is a Kconfig option for SLAAC:
https://github.com/zephyrproject-rtos/zephyr/blob/c99392f2d6bce1bfe16444bcc86e03a04ff0fa73/subsys/net/l2/openthread/Kconfig#L204

lualven-duplicate, 2020-03-26:
ok, that's good.

closed 2020-03-26 by lualven-duplicate