zephyrproject-rtos/zephyr #23712: Error in mounting the SD card

jagrutik9, 2020-03-24:
Hello,
I am working on nRF9160DK using the ncs tag v1.2.0
Zephyr version is v2.1.99
I am trying this example zephyr/samples/subsys/fs/fat_fs with nRF9160DK.
It always shows this error:
`***** Booting Zephyr OS build v2.1.99-ncs1-snapshot2-2647-gd6e67554cfeb *****
[00:00:00.024,078] [0m<inf> spi_nrfx_spim: CS control inhibited (no GPIO device)[0m
[00:00:05.039,306] [1;31m<err> main: Storage init ERROR![0m
Error mounting disk.
[00:00:10.054,565] [1;31m<err> fs: fs mount error (-5)[0m`

In the documentation I read that
`This sample can be built for an nrf52840_blip board. It requires both the nrf52840_blip.overlay and the dts_fixup.h for nrf52840_blip to work:`
So is it possible to use this example on nRF9160DK as it uses zephyr rtos?
If yes, are there any changes required so that it will work on nRF9160DK?
I have added overlay file also for nRf91.
nrf9160_pca10090ns.overlay:
`&spi3 {
        status = "okay";
        cs-gpios = <&gpio0 16 0>;

        sdhc0: sdhc@0 {
                compatible = "zephyr,mmc-spi-slot";
                reg = <0>;
                status = "okay";
                label = "SDHC0";
                spi-max-frequency = <24000000>;
        };
};

&spi3 {
        status = "okay";
        sck-pin = <19>;
        mosi-pin = <18>;
        miso-pin = <17>;
};`

ntavish, 2020-03-26:
Could it be an issue with initialization priorities? `disk_access_spi_sdhc.c` priority is 40, while SPI driver in `drivers/spi/spi_nrrfx_spim.c` priority is 70. Could you test that, and post the results, maybe we should change default priorities here.

jagrutik9, 2020-04-01:
I followed this thread https://devzone.nordicsemi.com/f/nordic-q-a/59430/nrf9160-sd-card-access-via-spi/242163#242163 on devzone forum and added CONFIG_SPI_NRFX_RAM_BUFFER_SIZE=8 in prj.conf file and it is working. SD card is mounted.

carlescufi, 2020-04-07:
Closing since it has been solved.

closed 2020-04-07 by carlescufi