zephyrproject-rtos/zephyr #23713: CMake integration with libmetal errors-out with the bleeding edge CMake release

SebastianBoe, 2020-03-24:
**Describe the bug**
Building a libmetal application with the newly released CMake version 3.17.0 will trigger a build error.

**To Reproduce**
Steps to reproduce the behavior:
1. cmake -DBOARD=nrf5340_dk_nrf5340_cpuapp -Hzephyr/samples/bluetooth/peripheral_hr -Boe && ninja -C oe
4. See error:

`ninja: error: 'cmake_object_order_depends_target_metal_', needed by 'cmake_object_order_depends_target_zephyr_final_', missing and no known rule to make it
`

**Impact**
Users must not use the latest CMake version if they are to use libmetal.

Does not affect 3.16.x.

SebastianBoe, 2020-03-24:
This is likely due to CMake behaving slightly differently in CMake 3.17.x and us not having adjusted our build scripts to this new behaviour yet.

It appears to have something to do with how CMAKE_BUILD_TYPE is manipulated by the libmetal build scripts.

SebastianBoe, 2020-03-24:
I have a fix now, will push soon.

tejlmand, 2020-08-06:
Has been fixed on a later upmerge of open-amp

closed 2020-08-06 by tejlmand