zephyrproject-rtos/zephyr #23723: Poor sinf/cosf performance compared to ARM CMSIS

too1, 2020-03-24:
The performance of the sinf/cosf functions in NEWLIB on Zephyr seems very poor compared to the ARM CMSIS implementation used in the nRF5 SDK for the nRF52 series. 
Preliminary numbers (until I have time to do a more controlled test) imply a runtime of around 15us on the nRF53 @ 128MHz, vs ~2.5us on the nRF52 @ 64MHz. 

An interesting aspect is that replacing cosf with cos only leads to a 30-40% increase in runtime, which indicates that the cosf implementation doesn't properly utilize the floating point unit in the nRF53 (32-bit floating point with HW acceleration should be significantly faster than 64-bit float without HW acceleration). 

I will update this issue as soon as I have time to do a more controlled test.

stephanosio, 2020-03-24:
The newlib `libm` implementation is in C and there is currently no ARM-optimised (asm) implementation available (hence, slow performance).

I expect these math performance issues to be mostly addressed by adding the CMSIS-DSP library support (I am currently working on this; there is a preliminary PR for it too: #21600).

stephanosio, 2020-03-24:
If you can, please try benchmarking against `arm_sin_f32` and `arm_cos_f32` in the PR; `tests/lib/cmsis_dsp/fastmath` would be a good base to use.

stephanosio, 2020-03-24:
> which indicates that the cosf implementation doesn't properly utilize the floating point unit

Please check/post the asm listing of the `sinf` and `cosf` functions.

too1, 2020-03-31:
I tried using PR #21600, but I don't see improved performance (actually it is slightly slower, 3.0/40.6ms vs 2.96/39.6ms). 

I use this prj.conf, with floating point enabled:
CONFIG_GPIO=y
CONFIG_NEWLIB_LIBC=y 

.map file added for reference:
[zephyr_map.zip](https://github.com/zephyrproject-rtos/zephyr/files/4409034/zephyr_map.zip)

Do I need to do something to enable the new math libraries?

stephanosio, 2020-04-01:
@too1 When running performance tests, please make sure that all operational variables are declared `volatile` so that the compiler doesn't try anything funny (e.g. nuking them).

I have tested newlib `cosf` and CMSIS-DSP `arm_cos_f32` functions and I see quite a significant performance difference between them:

|   input  | cosf<br/>(newlib) | arm_cos_f32<br/>(CMSIS-DSP) |
|:--------:|:-------------:|:-----------------------:|
|  2 * pi  |   1,408,960   |         577,594         |
| 200 * pi |   15,040,964  |         578,479         |

_(in cycles, measured with DWT on Cortex-M7; ATSAME70 Rev. B)_

Test code: https://gist.github.com/stephanosio/3a2de5e7f192acd5860d9ce866e1f2ba

You can also checkout the following test branch and run the modified `samples/hello_world`:
https://github.com/stephanosio/zephyr/tree/cmsis_dsp_perf_test/samples/hello_world

stephanosio, 2021-09-22:
Closing this since the relatively slow performance of the trigonometric functions provided by the newlib (libm) on ARM is really an upstream newlib issue and not a Zephyr issue.

A workaround has been provided through the CMSIS-DSP library, which provides the optimised high performance trigonometric function implementations for the ARM targets.

closed 2021-09-22 by stephanosio