zephyrproject-rtos/zephyr #23763: net: sockets: Wrong binding when connecting to ll address

lbise, 2020-03-25:
I think I found a bug when using the sockets. If you try to connect to a link local address but your device has more than one interface the net context will always be bound to the default interface. Therefore if the target is actually on the second interface the connection attempts will actually be made on the wrong interface.

This is because during the connect call at some point net_context_connect will call bind_default which will call net_context_bind to bind the net context. Then net_if_ipv6_select_src_iface is called to determine on which interface to bind. The call to net_if_ipv6_select_src_addr will return an unspecified address which leads to binding to the default interface because it cannot match using the IP address.

jukkar, 2020-03-25:
What network technology you are using, Ethernet or something else?

lbise, 2020-03-27:
I was using Ethernet.

jukkar, 2020-03-29:
> If you try to connect to a link local address but your device has more than one interface the net context will always be bound to the default interface.

This is actually expected if user has not called bind(). There is no way to know in which interface the target ll address is found (if it is not in neighbor cache already). So if the user has not called bind() to set the local end point which selects the network interface, then we are forced to select the default interface (as it is as good guess as any other way to select a network interface).
So in order this to work as expected, one need to call bind() to set the sending interface.

closed 2020-03-29 by jukkar