zephyrproject-rtos/zephyr #23777: Problem with applying overlay for custom board in blinky example

kubazdz, 2020-03-25:
**Description**
I am trying to flash my custom board with blinky sample. I have created nrf52811_pca10056.overlay file. My board is based on nrf52811 and the overlay file is made only to switch led0 pin to 18 (13 originally). After trying to build the example, overlay is loaded and the "Unable to parse input tree" error is risen due to the syntax error in nrf52811_pca10056.dts.pre.tmp file pointing at line with this code:

`gpios = <&gpio0 18 GPIO_ACTIVE_LOW>;`

need help to figure out what is wrong or is it some bug

**To Reproduce**
Steps to reproduce the behavior:
1. add nrf52811_pca10056.overlay as follows:
```
/{
	led0 {
		gpios = <&gpio0 18 GPIO_ACTIVE_LOW>;
		label = "Green LED 0";
	};
};
```
2. run 
`west build -b nrf52_pca10056`

**Expected behavior**
Succesful build of blinky sample with led0 pin changed to 18

**Impact**
Can't build an app

**Screenshots or console output**
[image]

**Environment (please complete the following information):**
 - OS: Ubuntu 18.04 LTS
 - Toolchain Zephyr SDK

carlescufi, 2020-03-25:
Can you check if the PR  below fixes it?

https://github.com/zephyrproject-rtos/zephyr/pull/23609

kubazdz, 2020-03-26:
@carlescufi it doesn't fix the problem, nothing changes

carlescufi, 2020-04-07:
I just tested this and it worked fine:

```
carles@zephyr-nrf ~/src/zephyr/zephyr/samples/basic/blinky (master)
$ cat boards/nrf52840dk_nrf52811.overlay
/{
        led0 {
                gpios = <&gpio0 18 GPIO_ACTIVE_LOW>;
                label = "Green LED 0";
        };
};

carles@zephyr-nrf ~/src/zephyr/zephyr/samples/basic/blinky (master)
$ west build -b nrf52840dk_nrf52811 .
-- west build: generating a build system
Including boilerplate (Zephyr base): /home/carles/src/zephyr/zephyr/cmake/app/boilerplate.cmake
-- Application: /home/carles/src/zephyr/zephyr/samples/basic/blinky
-- Zephyr version: 2.2.99 (/home/carles/src/zephyr/zephyr)
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.7.5", minimum required is "3.6")
-- Board: nrf52840dk_nrf52811
-- Found west: /home/carles/.local/bin/west (found suitable version "0.7.2", minimum required is "0.7.1")
-- Found toolchain: zephyr (/home/carles/bin/zephyr-sdk)
-- Found BOARD.dts: /home/carles/src/zephyr/zephyr/boards/arm/nrf52840dk_nrf52811/nrf52840dk_nrf52811.dts
-- Found devicetree overlay: /home/carles/src/zephyr/zephyr/samples/basic/blinky/boards/nrf52840dk_nrf52811.overlay
-- Generated zephyr.dts: /home/carles/src/zephyr/zephyr/samples/basic/blinky/build/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: /home/carles/src/zephyr/zephyr/samples/basic/blinky/build/zephyr/include/generated/devicetree_unfixed.h
Parsing /home/carles/src/zephyr/zephyr/Kconfig
Loaded configuration '/home/carles/src/zephyr/zephyr/boards/arm/nrf52840dk_nrf52811/nrf52840dk_nrf52811_defconfig'
Merged configuration '/home/carles/src/zephyr/zephyr/samples/basic/blinky/prj.conf'
Configuration saved to '/home/carles/src/zephyr/zephyr/samples/basic/blinky/build/zephyr/.config'
Kconfig header saved to '/home/carles/src/zephyr/zephyr/samples/basic/blinky/build/zephyr/include/generated/autoconf.h'
-- The C compiler identification is GNU 9.2.0
-- The CXX compiler identification is GNU 9.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/carles/bin/zephyr-sdk/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc
-- Cache files will be written to: /home/carles/.cache/zephyr
-- Configuring done
-- Generating done
-- Build files have been written to: /home/carles/src/zephyr/zephyr/samples/basic/blinky/build
-- west build: building application
[1/124] Preparing syscall dependency handling

[119/124] Linking C executable zephyr/zephyr_prebuilt.elf
Memory region         Used Size  Region Size  %age Used
           FLASH:       11916 B       192 KB      6.06%
            SRAM:        3848 B        24 KB     15.66%
        IDT_LIST:          56 B         2 KB      2.73%
[124/124] Linking C executable zephyr/zephyr.elf
```

carlescufi, 2020-04-07:
@anangl or @mbolivar any ideas about what could be wrong here?

mbolivar-nordic, 2020-04-07:
> 1. add nrf52811_pca10056.overlay as follows:
> 
> ```
> /{
> 	led0 {
> 		gpios = <&gpio0 18 GPIO_ACTIVE_LOW>;
> 		label = "Green LED 0";
> 	};
> };
> ```
> 
> 1. run
>    `west build -b nrf52_pca10056`

The board names in steps 1 and 2 do not match. Does the problem persist even when the overlay is for the right board?

anangl, 2020-04-08:
@kubazdz The flag `GPIO_ACTIVE_LOW` is not available in Zephyr 2.1.0. Either use `GPIO_INT_ACTIVE_LOW` instead or upgrade to Zephyr 2.2.0.

kubazdz, 2020-04-08:
Right, thanks all, I managed to apply all your comments and to build  exapmle. It was all matter of a wrong version

kubazdz, 2020-04-08:
actually this issue can be closed, but I have another similar to this but with few changes. I made a new issue 

#24193

mbolivar-nordic, 2020-04-08:
> actually this issue can be closed, but I have another similar to this but with few changes. I made a new issue

Thanks! We'll continue discussion in the new issue then.

closed 2020-04-08 by mbolivar-nordic