zephyrproject-rtos/zephyr #23798: UART driver API should be made more power-friendly

vanti, 2020-03-26:
**Is your enhancement proposal related to a problem? Please describe.**
Per discussion in #22998, currently the UART driver API exposes uart_poll_in() and uart_poll_out() which can be called at any time to check for input or output character, with the assumption that the UART remains active outside of the function call. This essentially means the UART cannot be automatically powered-down in the lifetime of the application (e.g. by the PM policy using the central method), short of having the application manually managing it.

**Describe the solution you'd like**
The API should be refined to be more power-friendly. 

**Describe alternatives you've considered**
Some possible solutions:
1. Remove the assumption by doing the polling inside uart_poll_in() and uart_poll_out(), such that uart_poll_in() would only pick up input while inside the function call, and uart_poll_out() would busy-wait for output to be complete.
2. Drop support for these poll-based APIs and use interrupt-driven API instead, where we can for example disable the uart when the uart irq is disabled.

jenmwms, 2020-03-26:
In #2 solution in the description, would this require non-interrupt-based designs to then implement interrupts to comply with the API? Perhaps it only effects a small audience, but it's worth mentioning that there are some designs that choose to avoid interrupts and this would remove that choice?

vanti, 2020-03-26:
> In #2 solution in the description, would this require non-interrupt-based designs to then implement interrupts to comply with the API? Perhaps it only effects a small audience, but it's worth mentioning that there are some designs that choose to avoid interrupts and this would remove that choice?

@jenmwms Yes it would require non-interrupt-based designs to implement interrupts. If Zephyr has support for designs that choose to avoid interrupts, then this is not a viable solution.

I have added solution #3 which may be more appropriate.

tautologyclub, 2020-04-08:
Well let the app developer do as she wishes, but zephyr components using poll_in should definitely 100% be discouraged imo.

carlescufi, 2020-04-21:
@pabigot FYI, since you've been working on this.