zephyrproject-rtos/zephyr #23799: tests/subsys/logging/log_immediate failed on reel_board

chen-png, 2020-03-26:
**Description**
Running tests/subsys/logging/log_immediate on reel_board, there are many error log output.

**To Reproduce**
Steps to reproduce the behavior:
1. west build -b reel_board -p auto tests/subsys/logging/log_immediate/
2. west flash
3. See error

**Screenshots or console output**
*** Booting Zephyr OS build zephyr-v2.2.0-622-ge27cf1576338  ***
Running test suite test_log_immediate

starting test - test_log_immediate_preemption
[00:00:00.028,167] [0m<inf> test: test string printed 1 2 0x20000008[0m
[00:00:00.034,729] [0m<inf> test: data:
00 00 00 00 00 00 00 00                          |........         [0m
[00:00:00.044,799] [0m<inf> test: test string printed 1 2 0x2000009c[0m
[00:00:00.051,361] [0m<inf> test: data:
00 20 00 20 50 0e 00 20  60 0e 00 20 65 4f 00 00 |. . P..  `.. eO..[0m
[00:00:00.061,492] [0m<inf> test: test string printed 1 2 0x20000130[0m
[00:00:00.068,176] [0m<inf> test: data:
00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 |........ ........
00 00 00 00 00 00 00 00                          |........         [0m
[[00:00:00.082,580] [0m<inf> test: test string printed 1 2 0x2000009c[0m
[00:00:00.091,400] [0m<inf> test: data:
00 20 00 20 50 0e 00 20  60 0e 00 20 65 4f 00 00 |. . P..  `.. eO..[0m
00:00:00.084,594] [0m<inf> test: test string printed 1 2 0x200001c4[0m
[00:00:00.108,245] [0m<inf> test: data:
00 00 00 80 20 00 00 00  20 00 00 00 20 00 00 00 |.... ...  ... ...
20 00 00 00 98 07 00 20  20 00 00 00 20 00 00 00 | ......   ... ...[0m
[00:00:00.124,633] [0m<inf> test: test string printed 1 2 0x20000258[0m
[00:00:00.131,317] [0m<inf> test: data:
20 00 00 00 20 00 00 00  98 07 00 20 00 00 22 09 | ... ... ... ..".
80 1a 00 20 18 2c 00 20  02 00 00 00 18 2c 00 20 |... .,.  .....,. 
14 2c 00 20 d5 ee 01 00                          |.,.[00:00:00.064,849] [0m<inf> test: test string printed 1 2 0x20000008[0m
[00:00:00.158,996] [0m<inf> test: data:
00 00 00 00 00 00 00 00                          |........         [0m

**Environment (please complete the following information):**
 - OS: fedora28
 - Toolchain: zephyr-sdk-0.11.1
 - Commit ID: e27cf157633

nashif, 2020-03-31:
bisected and found this to be the bad commit: 0ca40205e56f7192c12dccdb2c25106b0b2bc5ae

@Mierunski FYI

nashif, 2020-04-01:
@carlescufi  fyi

carlescufi, 2020-04-09:
@nordic-krch and @Mierunski can you take a look and submit a PR? If you are unable to at this time please let me know.

nashif, 2020-05-21:
@nordic-krch  @Mierunski  @carlescufi  ping!

nordic-krch, 2020-05-21:
This is a valid behavior (test is passing). see #20902 description:

```
Changed behavior of the logger in immediate mode. Previously, context was lock for whole string processing and sending which resulted in log output being clean and not interleaved but at the cost of very long period of interrupt locking. It is changed to lock interrupts only for single byte sending over transport (e.g. uart). With this approach long locking time is removed but log output may be garbled since one log message can be interrupted by another.
```

Test just checks the stability. What about changing text that is printed to contain something like `potentially garbled` to avoid misinterpretation.

carlescufi, 2020-05-22:
This is actually a real bug in the  UART or the logging subsystem. 
The reason the test fails is that the completion banner on reel_board is never printed:
```
PASS - test_log_immediate_preemption
===================================================================
Test suite test_log_immediate succeeded
===================================================================
PROJECT EXECUTION SUCCESSFUL
```
The problem is that without enabling `CONFIG_LOG_IMMEDIATE_CLEAN_OUTPUT` the test never reaches the end, instead being stuck in `uart_poll_out()`:

```
#0  arch_swap (key=key@entry=0) at /home/carles/src/zephyr/zephyr/arch/arm/core/aarch32/swap.c:60
#1  0x00005d52 in z_swap_irqlock (key=0) at ../kernel/include/kswap.h:175
#2  z_swap (key=..., lock=0x2000225c <z_main_stack+220>) at ../kernel/include/kswap.h:175
#3  z_tick_sleep (ticks=33) at /home/carles/src/zephyr/zephyr/kernel/sched.c:1185
#4  z_tick_sleep (ticks=33) at /home/carles/src/zephyr/zephyr/kernel/sched.c:1145
#5  0x00005e00 in z_impl_k_sleep (timeout=...) at /home/carles/src/zephyr/zephyr/kernel/sched.c:1215
#6  0x000034ac in k_sleep (timeout=...) at zephyr/include/generated/syscalls/kernel.h:127
#7  k_msleep (ms=1) at ../include/kernel.h:911
#8  uart_nrfx_poll_out (dev=<optimized out>, c=<optimized out>)
    at /home/carles/src/zephyr/zephyr/drivers/serial/uart_nrfx_uart.c:281
#9  0x00000f5a in z_impl_uart_poll_out (out_char=61 '=', dev=<optimized out>) at ../include/drivers/uart.h:640
#10 uart_poll_out (out_char=61 '=', dev=<optimized out>) at zephyr/include/generated/syscalls/uart.h:112
#11 console_out (c=61) at /home/carles/src/zephyr/zephyr/drivers/console/uart_console.c:93
#12 0x00000630 in z_vprintk (out=out@entry=0x54d <char_out>, ctx=ctx@entry=0x20002314 <z_main_stack+404>,
    fmt=0x85ab "=======\n", fmt@entry=0x856f '=' <repeats 67 times>, "\n", ap=...)
    at /home/carles/src/zephyr/zephyr/lib/os/printk.c:118
#13 0x00000942 in vprintk (fmt=fmt@entry=0x856f '=' <repeats 67 times>, "\n", ap=..., ap@entry=...)
    at /home/carles/src/zephyr/zephyr/lib/os/printk.c:352
#14 0x00006c10 in printk (fmt=0x856f '=' <repeats 67 times>, "\n")
    at /home/carles/src/zephyr/zephyr/lib/os/printk.c:405
#15 0x000021c2 in run_test (test=0x20000000 <_test_log_immediate.9370>)
    at /home/carles/src/zephyr/zephyr/subsys/testsuite/ztest/src/ztest.c:340
#16 z_ztest_run_test_suite (name=name@entry=0x7ab5 "test_log_immediate",
    suite=suite@entry=0x20000000 <_test_log_immediate.9370>)
    at /home/carles/src/zephyr/zephyr/subsys/testsuite/ztest/src/ztest.c:361
#17 0x00000544 in test_main ()
    at /home/carles/src/zephyr/zephyr/tests/subsys/logging/log_immediate/src/log_immediate_test.c:82
#18 0x00006f94 in main () at /home/carles/src/zephyr/zephyr/subsys/testsuite/ztest/src/ztest.c:425
```

PR #25534 (2020-05-25): tests: logging: Add a clean ouput test and clarify output
Add a new test with clean output and warn the user that the version
without clean output will output garbled text.

Fixes #23799
commit 855b6c1a (2020-05-25): tests: logging: Add a clean ouput test and clarify output

Add a new test with clean output and warn the user that the version
without clean output will output garbled text.

Fixes #23799
closed 2020-05-25 by carlescufi