zephyrproject-rtos/zephyr #23800: tests/drivers/counter/counter_cmos failed on acrn platform

chen-png, 2020-03-26:
**To Reproduce**
Steps to reproduce the behavior:
1. sanitycheck -p acrn --device-testing --device-serial /dev/ttyUSB1 
    -T tests/drivers/counter/counter_cmos -west-flash="acrn_up2.sh"
2. See error

**Screenshots or console output**

*** Booting Zephyr OS build zephyr-v2.2.0-622-ge27cf1576338  ***
Running test suite test_cmos_counter

starting test - test_cmos_rate
**Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/counter/counter_cmos/src/main.c:36: test_cmos_rate: (elapsed >= MIN_BOUND is false)
busted minimum bound**
FAIL - test_cmos_rate

Test suite test_cmos_counter failed.

PROJECT EXECUTION FAILED

**Environment (please complete the following information):**
 - OS: fedora28
 - Toolchain : zephyr-sdk-0.11.1
 - Commit ID: e27cf157633

tbursztyka, 2020-04-28:
I raised the delay of the test, added some printouts, and tried many times. I got it to work only once...
So the vm RTC clock seems to be incremented, only too slowly?

> *** Booting Zephyr OS build zephyr-v2.2.0-1980-gba9f20a6ebbb  ***
Running test suite test_cmos_counter
===================================================================
starting test - test_cmos_rate
Start value 1588066051
Elapsed value 1588066052
difference: 1
PASS - test_cmos_rate
===================================================================
Test suite test_cmos_counter succeeded
===================================================================
PROJECT EXECUTION SUCCESSFUL

I restarted soon after:

> *** Booting Zephyr OS build zephyr-v2.2.0-1980-gba9f20a6ebbb  ***
Running test suite test_cmos_counter
===================================================================
starting test - test_cmos_rate
Start value 1588066157
Elapsed value 1588066157
difference: 0

    Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/counter/counter_cmos/src/main.c:42: test_cmos_rate: (elapsed >= MIN_BOUND is false)
busted minimum bound
FAIL - test_cmos_rate
===================================================================
Test suite test_cmos_counter failed.
===================================================================
PROJECT EXECUTION FAILED

So the clock advanced of 2 minutes between the test which seems right compared to the actual time it took me to kill the vm and restart it.

I changed the k_msleep time then to 60000 (60 seconds)... and it did not slept for 60 seconds. Barely 10. 

Looks like there is something wrong with the internal timer, not the cmos counter. That could explain the other issues #23801 and #23802

tbursztyka, 2020-04-28:
Seems to work fine if option `--lapic_pt` is given to acrn-dm (in launcher script).

However, the VM is then unstoppable (even with -f)... Had to reboot the hypervisor. But that's another issue.

@chen-png Can you retry with adding `--lapic_pt` to your script that runs the vm?

chen-png, 2020-04-29:
@tbursztyka  i added the --lapic_pt option after acrn-dm command as below:

acrn-dm -A -m $mem_size -s 0:0,hostbridge -s 1:0,lpc -l com1,stdio \
  -s 5,virtio-console,@pty:pty_port \
  -s 3,virtio-blk,./zephyr.img \
  --ovmf /usr/share/acrn/bios/OVMF.fd \
  --lapic_pt \
  $vm_name

but it seems that it couldn't start a vm successfully.

 DEVICE: [601668277373us][cpu=0][vm0:vcpu0][sev=3][seq=225]:Wrong guest flags 0x6
DEBUG   - DEVICE:
DEBUG   - DEVICE: [601668801384us][cpu=0][vm0:vcpu0][sev=3][seq=226]:Wrong guest flags 0x6
DEBUG   - DEVICE:
DEBUG   - DEVICE: [601669309357us][cpu=0][vm0:vcpu0][sev=3][seq=227]:Wrong guest flags 0x6
DEBUG   - DEVICE:
DEBUG   - DEVICE: [601669817348us][cpu=0][vm0:vcpu0][sev=3][seq=228]:Wrong guest flags 0x6
DEBUG   - DEVICE:
DEBUG   - DEVICE:
DEBUG   - DEVICE: ACRN:\>vm_consol[601670325328us][cpu=0][vm0:vcpu0][sev=3][seq=229]:Wrong guest flags 0x6

can't find zephyr_vm1
try it again...
can't find zephyr_vm1
can't find zephyr_vm1
stop vm fail.

tbursztyka, 2020-05-12:
Which version of ACRN? I am using the latest packaged in clear linux.
You did not forget to do the post-processing on zephyr image to insert grub, did you?

chen-png, 2020-05-13:
i used the ACRN Branch : v1.5, ACRN Commit ID : 6c554faab3.
yes, i did, the zephyr image process was done by a script automatically, it worked for other tests.

tbursztyka, 2020-05-15:
Ok switch to 1.6, this works on this version.

carlescufi, 2020-05-21:
@nashif @tbursztyka any chance we can close this in time for 2.3.0?

tbursztyka, 2020-05-27:
@carlescufi  I think you can close. it's not a bug in zephyr, just an option that needs to be used in acrn-dm side.

carlescufi, 2020-05-27:
Closing as per @tbursztyka's comments

closed 2020-05-27 by carlescufi

chen-png, 2020-06-19:
@tbursztyka 
for this test, I updated acrn version to v1.6(commit: 9b6fa69552), **and added --lapic_pt option after acrn-dm command, it seems that it couldn't start a vm successfully**, do you know why?

test results:
 ACRN:\>vm_con[127053209us][cpu=3][vm0:vcpu3][sev=3][seq=46]:vlapic: vm_apicid2vcpu_id: bad lapicid 4
DEBUG   - DEVICE: sole 1
DEBUG   - DEVICE: VM is not valid
DEBUG   - DEVICE:
DEBUG   - DEVICE: Error: Invalid parameters.
DEBUG   - DEVICE: **ACRN:\>[128547553us][cpu=0][vm0:vcpu0][sev=3][seq=48]:Wrong guest flags 0x6**
DEBUG   - DEVICE:
DEBUG   - DEVICE: [129070377us][cpu=0][vm0:vcpu0][sev=3][seq=49]:Wrong guest flags 0x6
DEBUG   - DEVICE:
DEBUG   - DEVICE: [129577591us][cpu=0][vm0:vcpu0][sev=3][seq=50]:Wrong guest flags 0x6
DEBUG   - DEVICE:
DEBUG   - DEVICE: [130084808us][cpu=0][vm0:vcpu0][sev=3][seq=51]:Wrong guest flags 0x6
DEBUG   - DEVICE:
DEBUG   - DEVICE: [130592022us][cpu=0][vm0:vcpu0][sev=3][seq=52]:Wrong guest flags 0x6
DEBUG   - DEVICE:
DEBUG   - DEVICE: [131099246us][cpu=0][vm0:vcpu0][sev=3][seq=53]:Wrong guest flags 0x6
DEBUG   - DEVICE:
DEBUG   - DEVICE: [131606472us][cpu=0][vm0:vcpu0][sev=3][seq=54]:Wrong guest flags 0x6
DEBUG   - DEVICE:
DEBUG   - DEVICE: [132114593us][cpu=0][vm0:vcpu0][sev=3][seq=55]:Wrong guest flags 0x6
DEBUG   - DEVICE: