zephyrproject-rtos/zephyr #23803: nrf52840 ble error

Narendhar95559, 2020-03-26:
Hi,

Iam using nrf52840 development board, developing software using zephyr.
In software development iam using bluetooth as a peripheral and uart in interrupt mode(to get data from other module), when iam trying to run both interfaces the getting below errors and iam included watchdog timer so after sometime it is restarting.

[00:01:36.902,862] <inf> os: prio recv thread stack : unused 336 usage 112 / 448 (25 %)
[00:01:36.902,893] <err> os: ***** BUS FAULT *****
[00:01:36.902,893] <err> os:   Imprecise data bus error
[00:01:36.902,923] <err> os: r0/a1:  0x00000029  r1/a2:  0x00000004  r2/a3:  0x00000000
[00:01:36.902,923] <err> os: r3/a4:  0x20004a0c r12/ip:  0x00000000 r14/lr:  0x000388fb
[00:01:36.902,923] <err> os:  xpsr:  0x61000000
[00:01:36.902,923] <err> os: Faulting instruction address (r15/pc): 0x000390ee
[00:01:36.902,923] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
[00:01:36.902,923] <err> os: Current thread: 0x20000eec (unknown)
[00:01:37.159,637] <err> os: Halting system

When i checked with above address error is at
zephyr/include/sys/dlist.h:211

I tried to enable stack usage but iam getting usage prints for only advertising start
[00:00:00.990,539] <inf> os: ecc stack :	unused 236	usage 864 / 1100 (78 %)

Not getting usage for connect,read,write,disconnect

Narendhar95559, 2020-03-26:
I enabled below flags
CONFIG_INIT_STACKS=y
CONFIG_THREAD_MONITOR=y
CONFIG_THREAD_STACK_INFO=y

Narendhar95559, 2020-04-13:
Can some one help me with above problem.

cvinayak, 2020-04-17:
The issue is suspected to be insufficient thread stack allocation.

> iam trying to run both interfaces

Depending on what you have configured your application to use UART, UART could be used for console apart from your indented use. Also, any code you write (auto variable declarations, for example) in a thread needs appropriate thread stack allocations.

@Narendhar95559 Please provide details of reproducing your issue, like, the sample application used, toolchain used, host operating system, steps used to build and test (Use the issue template made available when you created the github issue).

You will have to share a minimal application that reproduces your issue, i.e. if your issue pertains to new application developed by you. Preferably, if you can provide a diff of the failing application forked from an existing sample in the Zephyr repo, it will speed up analysis.

Narendhar95559, 2020-04-17:
Thanks for reply, at present issue is solved

As you mentioned the issue is insufficient stack allocation, but in zephyr(v2.1.0) main stack sizes available are 512,2048 and 8192 but for my application 4096 required.

I tried with latest zephyr(2.2.29) which had support for 4096 stack size.

closed 2020-04-18 by cvinayak