zephyrproject-rtos/zephyr #23808: ARM bus fault with code coverage

kkiyota, 2020-03-26:
My code coverage ELF has a bus fault.  Debugging with it, it occurred during while loop in the gcov_static_init as shown below.  My board is my own board wit ARM Cortex M7.

Could you let me know how I can adjust func_pointer_start(&__init_array_start) and func_pointer_end(&__init_array_end)?

So far, code coverage feature is unstable.  For example, without using a function in main, it works, but with using, it has the bus fault, etc.

>  

[image]

kkiyota, 2020-03-26:
Additional debugger info.

Thank you for your help!

[image]

andrewboie, 2020-03-26:
> Could you let me know how I can adjust func_pointer_start(&__init_array_start) and func_pointer_end(&__init_array_end)?

```
include/linker/common-rom.ld
43-	{
44-		. = ALIGN(4);
45:		__init_array_start = .;
46-		KEEP(*(SORT_BY_NAME(".init_array*")))
47:		__init_array_end = .;
48-	} GROUP_LINK_IN(ROMABLE_REGION)
49-#endif
```

kkiyota, 2020-03-26:
@andrewboie Thank you for your info!  

Could you also give me preferred examples since I'm not familiar with it?

Best regards,
Koji

kkiyota, 2020-03-27:
@andrewboie The section in common-rom.ld you post seems to be selected in case of "#ifdef CONFIG_CPLUSPLUS"

And this is with code coverage feature, I also found the following coverage_rodata.ld.

Which one should I use?  Working with AMR MPU, how can I fix my problem, usage fault?   

**_zephyr/subsys/testsuite/coverage/coverage_rodata.ld_**

```
/*
 * Copyright (c) 2019 Intel Corporation
 * SPDX-License-Identifier: Apache-2.0
 */

PROVIDE_HIDDEN (__init_array_start = .);
KEEP (*(SORT(.init_array.*)))
KEEP (*(.init_array*))
PROVIDE_HIDDEN (__init_array_end = .); 
```

kkiyota, 2020-03-31:
Looking at Kconfig.zephyr, in case of COVERAGE, "-O0" option for gcc is selected as shown below.  It doesn't work for my own board which is Cortex-M7 with MPU.  That is, there are usage fault caused by unaligned access.  This means that gcc generates unaligned instructions.  On the other hand, I specified "-O2" instead, it worked and generated coverage report. 

Does anyone know if this is gcc problem or exceptional case with MPU?

[image]

kkiyota, 2020-04-01:
In addition, ALIGN(4) needs to be added in my case as shown below.

[image]

andrewboie, 2020-04-02:
> In addition, ALIGN(4) needs to be added into zephyr/subsys/testsuite/coverage/coverage_rodata.ld in my case as shown below.

Does this solve your problem? Can you send a PR? It's odd that this isn't 4-byte aligned already, although I'd have to look at where this linker script fragment gets included

closed 2020-07-15 by github-actions[bot]