zephyrproject-rtos/zephyr #23824: ARM Cortex-M7 MPU setting

kkiyota, 2020-03-26:
Looking at my current MPU setting thru debugger feature, MPU Region 1 seems to be configured as XN="Not execute", MPU_RASR1=0x110B0027.  Currently, I don't explicitly specify it as XN="Not execute" anywhere.  

Could you let me know how I can change it to XN="Execute"?  If I can do it in *.dts, it would be preferable. 

> 

[image]

Thank you,
Koji

ioannisg, 2020-03-27:
Normally, Region 1 is the SRAM region (whole SRAM, or kernel SRAM in case of user mode). SRAM is set as XN, that's correct. This is done in ARM MPU initialization before kernel initialization. The SRAM is a "fixed" region that is configured at start-up. The configuration for SRAM is taken from the mpu region setting of the SoC (you need to see how the SoC specifies the fixed MPU regions to be configured at boot; look as the SoC definition directory). You can hack the setting for SRAM region to allow execution.

However, if you just want to execute something from SRAM, there is a Kconfig option for that - check ARCH_HAS_RAMFUNC_SUPPORT option.

kkiyota, 2020-03-28:
@ioannisg Thank you for your explanation and suggestion!  Although I applied "ARCH_HAS_RAMFUNC_SUPPORT", it still occurred.  

In further analysis, it may be related to cmake optimization for gcc.  It looks -O0?.  I'll try -O2 and others including "-mno-unaligned-access" tomorrow.  Any idea which one will be appropriate?

kkiyota, 2020-03-29:
@ioannisg "-O2" worked.  Any other suggestions?

ioannisg, 2020-03-30:
>> @ioannisg Thank you for your explanation and suggestion! Although I applied "ARCH_HAS_RAMFUNC_SUPPORT", it still occurred.

ARCH_HAS_RAMFUNC_SUPPORT simply indicates that Corte-xM is able to execute code from SRAM; setting it has no effect. (you can look at tests/arch/arm/arm_rumfunc for details of how to place code to SRAM)

>>In further analysis, it may be related to cmake optimization for gcc. It looks -O0?. I'll try -O2 and others including "-mno-unaligned-access" tomorrow. Any idea which one will be appropriate?

I cannot understand what _it_ refers to, and what the real problem it - I thought you're asking about the settings of MPU regions.

if you got your answers, or solved your issue, please, close this ticket.

kkiyota, 2020-03-30:
@ioannisg My original problem was described in https://github.com/zephyrproject-rtos/zephyr/issues/23808#issue-588334925 that I thought it was related to MPU setting.  So far, I believe it's caused by unaligned access which is caught by MPU as usage fault, and by code optimization by gcc.

Since there are several different code optimization options for gcc, I'm asking which options are appropriate other than "-O2".  It works so far.

If you have any idea, please let me know.  Otherwise, I'll close this one.

ioannisg, 2020-03-30:
-O2 should be the default, IIRC; i think O0 might be specific to when enabling code coverage.

kkiyota, 2020-03-30:
@ioannisg "-Og" was used in my usage fault case.

kkiyota, 2020-03-31:
Moved to https://github.com/zephyrproject-rtos/zephyr/issues/23824

closed 2020-04-02 by andrewboie