zephyrproject-rtos/zephyr #23825: edtlib.py fails to find bindings when DTS_ROOT is a relative path

crossboss, 2020-03-26:
**Describe the bug**
When building out-of-tree drivers, new bindings can also be located outside of the zephyr tree using `set(DTS_ROOT some_path)` in the application's CMakeList.txt. However, if `some_path` is a relative path (relative to the CMakeList.txt), the bindings are found, but can not be matched using the compatible string. In this case, no devicetree macros are generated.

**To Reproduce**
1. Add `set(DTS_ROOT ../my_driver/zephyr)` to the application CMakeList.txt
2. Create a binding file in ../my_driver/zephyr/dts/bindings/test.yaml
3. Add compatible string in test.yaml that reads "test-path"
4. In the devicetree file for your application, add a node with the same compatible string ("test-path")
5. Build the application

**Expected behavior**
Devicetree_unfixed.h should contain something like:
```
/*
 * Devicetree node:
 *   /test
 *
 * Binding (compatible = test-path):
 *  /home/user/my_driver/zephyr/dts/bindings/test.yaml
```

Instead it will read:
```
/*
 * Devicetree node:
 *   /test
 *

/* No macros: node has no matching binding */
```

**Impact**
This behaviour is unexpected and difficult to debug. It is also impractical to have to use absolute paths in DTS_ROOT, and makes out-of-tree driver development essentially non-portable between different machines.

**Environment:**
 - OS: MacOS
 - Toolchain: GNU 9.2.1 and west
 - Commit SHA: f090e2563a1de2773a0c142398e7d876dc84ab19

**Additional context**
Expected to behave the same on any OS

**Cause**
The problem is caused by the fact that the CWD when running the python script edtlib.py, is set to the build dir rather than the application dir. This causes the relative paths to be wrong when searching for the yaml files.

crossboss, 2020-03-27:
Maybe something like this in the application CMakeList.txt would solve the problem:
`set(DTS_ROOT ${CMAKE_CURRENT_SOURCE_DIR}/../my_driver/zephyr)`
But than a warning and some documentation would be nice

mbolivar-nordic, 2020-04-07:
CONF_FILE treats relative paths as relative to the application source directory:

```
set(merge_config_files_with_absolute_paths "")
foreach(f ${merge_config_files})
  if(IS_ABSOLUTE ${f})
    set(path ${f})
  else()
    set(path ${APPLICATION_SOURCE_DIR}/${f})
  endif()

  list(APPEND merge_config_files_with_absolute_paths ${path})
endforeach()
```

(from kconfig.cmake)

Perhaps DTS_ROOT should do the same?

Whatever we do, we should ensure consistent behavior with all the CMake _ROOT variables:

- BOARD_ROOT
- DTS_ROOT
- KCONFIG_ROOT (?)
- SOC_ROOT
- TOOLCHAIN_ROOT

arvid-r, 2020-04-07:
Exactly. I at least expected it to be relative to the application directory, which is true for BOARD_ROOT.

mbolivar-nordic, 2020-07-01:
> . Remove the stale label or add a comment saying that you would like to have the label removed otherwise this issue will automatically be closed in 14 days.

I removed the label. This is a real bug IMO.

tejlmand, 2020-07-09:
> Maybe something like this in the application CMakeList.txt would solve the problem:
> `set(DTS_ROOT ${CMAKE_CURRENT_SOURCE_DIR}/../my_driver/zephyr)`

Exactly.
Inside a CMakeLists.txt file, there is little reason to use a relative path without saying from where.
CMake already offers both `${CMAKE_CURRENT_LIST_DIR}` and `${CMAKE_CURRENT_SOURCE_DIR}`.

> CONF_FILE treats relative paths as relative to the application source directory:
>
> ```
> set(merge_config_files_with_absolute_paths "")
> foreach(f ${merge_config_files})
>   if(IS_ABSOLUTE ${f}) [...]

Good question, and another question is, is this principle safe to use this in general.
It actually becomes a tricky question when starting to think a little deeper about it.

### CMakeLists.txt files

Now, in #23825 having a single `CMakeLists.txt` then one can easily argue that it make sense to use path as relative to app dir.
But what if user has a bit more complex setup, and does:
```
- app-dir
  ├─ CMakeLists.txt
  └─ project_settings
         └─ CMakeLists.txt
- my_driver
   └─ dts
        └─ ...
```

and in the main `CMakeLists.txt` he does:
```
add_subdirectory(project_settings)
find_package(Zephyr)
```

and in `project_settings/CMakeLists.txt` user writes:
```
set(DTS_ROOT ../../my_driver PARENT_SCOPE)
```
would user then expect that he needed to use `../my_driver` or `../../my_driver` ?
In this particular case, the code is in a subfolder, and `add_subdirectory()` was used to add it, meaning scope has changed.

From how `target_sources()` interprets relative path:
> Relative source file paths are interpreted as being relative to the current source directory (i.e. CMAKE_CURRENT_SOURCE_DIR).

In the ROOT case we are setting a variable and not using it yet, which complicates matters a bit.

### cmake command line arguments

Roots can also be specified as CMake argument on command line.
The problem with `-DDTS_ROOT=<path>` as argument on command line, is that we cannot determine if user has been using `-DDTS_ROOT=<path>` as argument, or if `set(DTS_ROOT <path>)` was used inside another CMakeLists.txt.

Imagine following setup:
```
- applications
  └─ app1
        └─ CMakeLists.txt
- my_driver
   └─ dts
        └─ ...
- build
```
and user does:
```
$ cd build
$ cmake ... -DDTS_ROOT=../my_driver ../applications/app1
```
which seems logically, cause user refers relative to the directory from where he is invoking CMake.
But if we interprets it relative to app dir, then user should have used `../../my_driver` instead, which on command line is not that obviously.

Problem with cmake arguments is that we don't know where CMake was invoked, and hence we don't know relative to what ?

### Zephyr CMake Config Package

As a last possibility, user can also use the Zephyr CMake Config Package:
https://docs.zephyrproject.org/latest/guides/zephyr_cmake_package.html#zephyr-build-configuration-cmake-package

Now doing a relative path in there, what does that mean ?
Relative to the package, relative to the application using that package, or ?

### Proposal

My suggestion would be to disallow the use of relative path directly inside CMake, cause someone will get it wrong, and there are too many ways people can use the root wrongly.

Users has ways to ensure relative to something they decide.
In a CMakeLists.txt they can do:
```
set(DTS_ROOT ${CMAKE_CURRENT_SOURCE_DIR}/..)
or:
set(DTS_ROOT ${CMAKE_CURRENT_LIST_DIR}/..)
```

In a Zephyr CMake config package they can use:
```
set(DTS_ROOT ${CMAKE_CURRENT_LIST_DIR}/..)
```

On command line invocation they can do:
Windows
```
cmake -DDTS_ROOT=%CD%/../.. 
```
Linux:
```
cmake -DDTS_ROOT=$PWD/../.. 
```

My proposal is to check for relative path in CMake, and if a relative path is used, then issue an error, such as:
```
DTS_ROOT defined with relative path.
Please provide absolute path.

If DTS_ROOT was set in a CMakeLists.txt, you may use:
  set(DTS_ROOT ${CMAKE_CURRENT_LIST_DIR}/${DTS_ROOT})

if DTS_ROOT was specified as CMake argument, you may use:
  cmake -DDTS_ROOT=$PWD/${DTS_ROOT} (Linux/MacOS)
  cmake -DDTS_ROOT=%CD%/${DTS_ROOT} (Windows)
```

mbolivar-nordic, 2020-07-10:
> My proposal is to check for relative path in CMake, and if a relative path is used, then issue an error, such as:

I think this is a reasonable proposal.

tejlmand, 2020-09-08:
> 
> 
> > My proposal is to check for relative path in CMake, and if a relative path is used, then issue an error, such as:
> 
> I think this is a reasonable proposal.

Took a second lookat this, and found a way so that relative paths on command line, where `-D<some>_ROOT=<relative-path>` are allowed, and will be relative seen from application source dir.

If someone uses `set(<some>_ROOT <relative-path>)` then an error will be raised:

arvid-r, 2020-09-08:
Thanks. This seems like a good and safe solution.

PR #28180 (2020-09-10): Support relative path from application source dir
Fixes: #23825

Today, BOARD, SOC, ARCH, DTS, KCONFIG, and TOOLCHAIN_ROOT can be
specified by users or through other CMake files.

It is not clear if relative paths are permitted and/or from where a
relative path is constructed.

Inside CMake, it is very easy to specify `${ZEPHYR_BASE}`,
`${CMAKE_CURRENT_SOURCE_DIR}`, or similar, thus there is no reason to
use relative path inside CMake, as it easy can lead to discussion on
relative to what.

For users manually invoking CMake using, `cmake ... <app-dir>` it is
nice to have to possibility to spec [... cut]
commit 3083205f (2020-09-10): cmake: introduce zephyr_file() extension command

Fixes: #23825

This commit introduces a `zephyr_file()` extension command that allows
for examine a path list for relative paths, and convert relative paths
provided with `-D<path>` into absolute paths, starting from
`APPLICATION_SOURCE_DIR`.
commit fb16a3de (2020-09-10): cmake: ensure consistent behavior between x_ROOT settings

Fixes: #23825

Today, BOARD, SOC, ARCH, DTS, KCONFIG, and TOOLCHAIN_ROOT can be
specified by users or through other CMake files.

It is not clear if relative paths are permitted and/or from where a
relative path is constructed.

Inside CMake, it is very easy to specify `${ZEPHYR_BASE}`,
`${CMAKE_CURRENT_SOURCE_DIR}`, or similar, thus there is no reason to
use relative path inside CMake, as it easy can lead to discussion on
relative to what.

For users manually invoking CMake using, `cmake ... <app-dir>` it is
nice to have to possibilit [... cut]
closed 2020-09-10 by carlescufi