zephyrproject-rtos/zephyr #23850: Template with C linkage in util.h:52

javier-moreno-tridonic-com, 2020-03-27:
Maybe I can shed some light on the issue described in https://github.com/zephyrproject-rtos/zephyr/issues/9590 (closed by lack of information) here, as I'm currently facing a similar one if not exactly the same.

The problem is that this file defines a template if the user Is building a C++ application as I'm currently doing.

Someone (in my case, nordic hal) will include some of the zephyr headers within "extern C { }" blocks, thus, provoking a compilation error.

I my specific scenario. This is derived from the following inclusion trace:

```
In file included from /home/jagomo/code/myproject/zephyr/include/arch/arm/aarch32/cortex_m/memory_map.h:18,
                 from /home/jagomo/code/myproject/zephyr/include/arch/arm/aarch32/arch.h:37,
                 from /home/jagomo/code/myproject/zephyr/include/arch/cpu.h:19,
                 from /home/jagomo/code/myproject/zephyr/include/irq.h:15,
                 from /home/jagomo/code/myproject/modules/hal/nordic/./nrfx_glue.h:69,
                 from /home/jagomo/code/myproject/modules/hal/nordic/nrfx/nrfx.h:37,
                 from /home/jagomo/code/myproject/zephyr/soc/arm/nordic_nrf/nrf52/soc.h:16,
                 from /home/jagomo/code/myproject/zephyr/include/arch/arm/aarch32/cortex_m/cmsis.h:17,
                 from /home/jagomo/code/myproject/zephyr/include/arch/arm/aarch32/syscall.h:29,
                 from /home/jagomo/code/myproject/zephyr/include/arch/syscall.h:19,
                 from /home/jagomo/code/myproject/zephyr/include/syscall.h:12,
                 from /home/jagomo/code/myproject/zephyr/include/logging/log_core.h:14,
                 from /home/jagomo/code/myproject/zephyr/include/logging/log.h:11,
                 from ../../src/global-app/globalApp.cpp:12:
/home/jagomo/code/myproject/zephyr/include/sys/util.h:52:1: error: template with C linkage
   52 | template < class T, size_t N >
      | ^~~~~~~~
In file included from /home/jagomo/code/myproject/modules/hal/nordic/nrfx/nrfx.h:37,
                 from /home/jagomo/code/myproject/zephyr/soc/arm/nordic_nrf/nrf52/soc.h:16,
                 from /home/jagomo/code/myproject/zephyr/include/arch/arm/aarch32/cortex_m/cmsis.h:17,
                 from /home/jagomo/code/myproject/zephyr/include/arch/arm/aarch32/syscall.h:29,
                 from /home/jagomo/code/myproject/zephyr/include/arch/syscall.h:19,
                 from /home/jagomo/code/myproject/zephyr/include/syscall.h:12,
                 from /home/jagomo/code/myproject/zephyr/include/logging/log_core.h:14,
                 from /home/jagomo/code/myproject/zephyr/include/logging/log.h:11,
                 from ../../src/global-app/globalApp.cpp:12:
/home/jagomo/code/myproject/modules/hal/nordic/./nrfx_glue.h:36:1: note: 'extern "C"' linkage started here
   36 | extern "C" {
      | ^~~~~~~~~~
```

Being the key inclusion <irq.h> on nrfx_glue.h:69

If this inclusion is commented out. The project will build. But would be nice to not depend on this ad-hoc solution.

I was thinking about opening this issue on the HAL repository. But Since zephyr is written in C. Is not unlikely that people will add zephyr headers within "extern C" block. So probably is something that should be addressed also here. (Even if the addressing means a big red warning for "do not use "extern C" blocks in this fashion!" on the documentation)

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: zephyr-sdk.
 - Commit SHA or Version used
cat VERSION
VERSION_MAJOR = 2
VERSION_MINOR = 2
PATCHLEVEL = 0
VERSION_TWEAK = 0
EXTRAVERSION = rc3

javier-moreno-tridonic-com, 2020-03-27:
Just realized that on main.cpp I don't have this problem. Figured out that adding:
```
#include <zephyr.h>
```
before 
```
#include <logging/log.h>
```
Prevents completely the compilation error.

Is this the intended usage? Is so you can probably close this.

pabigot, 2020-03-30:
@carlescufi FYI This is a bug in the Nordic HAL.  Never ever should `#include` directives appear within an `extern` block.

carlescufi, 2020-04-07:
> @carlescufi FYI This is a bug in the Nordic HAL. Never ever should `#include` directives appear within an `extern` block.

@anangl can you please send a PR?

carlescufi, 2020-04-09:
This is done now, once https://github.com/zephyrproject-rtos/zephyr/pull/24062 is merged.

closed 2020-04-09 by carlescufi