zephyrproject-rtos/zephyr #23865: nrf52840 and pyocd cannot program at addresses above 512k

lawrence-king, 2020-03-27:
**Describe the bug**

Boards with nrf52840 chip using the pyocd debugger cannot program flash memory above the 512k mark. When using the command `west flash --hex-file foo.hex` this successfully programs a nrf52840 device using the pyocd runner, provided the contents of the file foo.hex are for addresses below the 512k mark. 

Here is an example hex file with content for above the 512k mark (foo.hex) which will fail to program:
```
:02000002F0000C
:107F0000A22EA1550960457CA4634314B00FCBC2D7
:107F1000DE96C473640F489EBD0007FFFFEFA7E025
:107F20007DF748B8AB3C4425A32C51CBA0C60EEA44
:107F3000D71257B80FA91E9A1F2441F637DC4135D6
:107F4000F86AC04EC3BC9ABC5ECAEE880379B4F02E
:107F500046FC24A2061475110B738A4D21CBEF59F0
:107F6000568CDBED956D8EAAFA14C72924D845509E
:107F7000C09D779DD7504626DEDE9B1148D1ABFBD6
:107F8000FF85FFEBD7642C3E29396D8F68D4A6504E
:107F9000D0ED6DC879F1895366AAF063B53E175BE1
:107FA000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFE1
:107FB000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFD1
:107FC000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFC1
:107FD000314847424834314A584D4E3130353535B5
:107FE00035FFFFFFFFFFFFFFFFFFFFFFFFFFFFFF6B
:107FF0001234567890123456789012345678905540
:00000001FF
```

**To Reproduce**
Steps to reproduce the behavior:
1. west build -b nrf52840_mdk
2. west flash --hex-file foo.hex

**Expected behavior**
The contents of foo.hex should be programmed into the device

**Impact**
This is an annoyance, but can be worked around, the challenge is the workaround is only needed for boards using nrf52840 and pyocd. Other nrf52840 boards using runner nrfjprog do not have this issue

**Screenshots or console output**
Here is the output in the failure case:
```
$ west flash --hex-file foo.hex 
-- west flash: rebuilding
[3/3] Completed 'project'
-- west flash: using runner pyocd
-- runners.pyocd: Flashing file: foo.hex
0000864:WARNING:file_programmer:Failed to add data chunk: no memory region defined for address 0x000f7f00
0000865:INFO:loader:Erased 0 bytes (0 sectors), programmed 0 bytes (0 pages), skipped 0 bytes (0 pages) at 0.00 kB/s
```

**Environment (please complete the following information):**
 - OS: Linux 19.04
 - Toolchain Zephyr SDK 0.11.2

**Additional context**
west flash runs the command `pyocd flash -e sector -a 0xc000 -t nrf52 foo.hex` which is incorrect, it should run the command `pyocd flash -e sector -a 0xc000 -t nrf52840 foo.hex`.

there are two possible workarounds:
1) manually run the pyocd command above to program foo.hex
2) call west flash with additional parameters `west flash --hex-file foo.hex -- --target=nrf52840`, however the additional parameters cause issues for boards that do not use pyocd as the runner.

I am not sure if this is a cmake issue, or a west issue.

SebastianBoe, 2020-03-27:
> west flash runs the command pyocd flash -e sector -a 0xc000 -t nrf52 foo.hex which is incorrect, it should run the command pyocd flash -e sector -a 0xc000 -t nrf52840 foo.hex.

FYI: @mbolivar-nordic

mbolivar-nordic, 2020-03-27:
This is a bug in the nrf52840_mdk/board.cmake file, not a general issue with nrf52840 boards as reported.

PR #23869 (2020-03-27): boards: nrf52840_mdk: fix pyocd integration
Fix the --target value.

Fixes: #23865
closed 2020-03-27 by galak