zephyrproject-rtos/zephyr #23874: Test case to check that kernel scrubs CPU registers of any sensitive data

maksimmasalski, 2020-03-27:
I'm developing a test case to check that kernel scrubs CPU registers of any sensitive data. 
I'm using a syscall and 0xDEADBEEF value, which should be written to registers, and after returning from system call into test code, as I expect that 0xDEADBEEF should be replaced with another value.
That will prove kernel scrubs CPU registers. 
**I made the system call, and right before return from the system call, I'm writing into the registers special value 0xDEADBEEF**

```
size_t z_impl_string_nlen(char *src, size_t maxlen, int *err)
{
    size_t ret;
    ret = z_user_string_nlen(src, maxlen, err);
#if defined(CONFIG_X86)
    __asm__ volatile (
		"movl $0xDEADBEEF, %ebx;\n\t"
		"movl $0xDEADBEEF, %ecx;\n\t"
		"movl $0xDEADBEEF, %edx;\n\t"
		//"movl $0xDEADBEEF, %esi;\n\t"
		//"movl $0xDEADBEEF, %edi;\n\t"
		//"movl $0xDEADBEEF, %esp;\n\t"
		//"movl $0xDEADBEEF, %ebp;\n\t"
		);
#elif defined(CONFIG_X86_64)
    __asm__ volatile (
		"movq $0xDEADBEEF, %rbx;\n\t"
		"movq $0xDEADBEEF, %rcx;\n\t"
		"movq $0xDEADBEEF, %rdx;\n\t"
		);
#elif defined(CONFIG_ARM)
    __asm__ volatile (
		"mov r2, #100;\n\t"
		);
#elif defined(CONFIG_ARC)
    __asm__ volatile (
		);
#else
#error "Not implemented for this architecture"
    zassert_unreachable("Write to control register fault");
#endif
	return ret;
}
```

**Then after exit from the system call in main test function, I check registers again for that value.**

```
void test_after_syscall_cpu_scrubs_regs(void)
{
	int err;
	size_t ret;
	ret = string_nlen(user_string, BUF_SIZE, &err);
#if defined(CONFIG_X86)
    int ebx_val;
    int ecx_val;
    int edx_val;
    __asm__ volatile (
            "\t movl %%ebx,%0" : "=r"(ebx_val));
    __asm__ volatile (
            "\t movl %%ecx,%0" : "=r"(ecx_val));
    __asm__ volatile (
            "\t movl %%edx,%0" : "=r"(edx_val));
    printk("\nReg ebx val user mode 0x%x\n", ebx_val);
    printk("\nReg ecx val user mode 0x%x\n", ecx_val);
    printk("\nReg edx val user mode 0x%x\n", edx_val);
#elif defined(CONFIG_X86_64)
    long int rbx_val;
    long int rcx_val;
    long int rdx_val;
    __asm__ volatile(
            "\t movq %%rbx,%0" : "=r"(rbx_val));
    __asm__ volatile(
            "\t movq %%rcx,%0" : "=r"(rcx_val));
    __asm__ volatile(
            "\t movq %%rdx,%0" : "=r"(rdx_val));
    printk("\nReg rbx val user mode 0x%lx\n", rbx_val);
    printk("\nReg rcx val user mode 0x%lx\n", rcx_val);
    printk("\nReg rdx val user mode 0x%lx\n", rdx_val);
#elif defined(CONFIG_ARM)
    int r0_val;
    __asm__ volatile (
            "\t mov %%r2,%0" : "=r"(r0_val));
#elif defined(CONFIG_ARC)
    __asm__ volatile (
		);
#else
#error "Not implemented for this architecture"
    zassert_unreachable("Write to control register fault");
#endif
}
```

**It is my rough demo, code can be dirty, but main idea should be clear. I'm not good at ARM, ARC registers, so does anybody knows what registers necessary to check and how to do that?** That test seems to be platform dependant, so different architectures are using different registers to store sensitive data. That test will verify that sensitive data won't remain in CPU's registers after returning from a system call.

stephanosio, 2020-03-28:
I think this test should be implemented in ASM (at least the `z_impl_string_nlen` and `test_after_syscall_cpu_scrubs_regs`); otherwise, there is no fundamentally safe way to guarantee that the regs will not get nuked by the compiler.

For x86, you should be able to DEADBEEF `esi` and `edi` as well; `esp` and `ebp` cannot be DEADBEEF'ed though since these regs are stack pointers. If you implement the test in ASM without using `ebp` to create stack frame, you will be able to DEADBEEF the `ebp` as well.

For x86-64, note that there are also additional GPRs (i.e. `r8-r15`).

For ARM, you should be able to DEADBEEF `r1`, `r2`, `r3` and `r12`. `r4` through `r11` are callee-saved registers and the compiler will ensure that they will be restored before the function returns anyway (even if you implement this test in ASM, you should restore them before returning from the function because the caller assumes that these regs will be preserved and may use them to, for example, store local variables.

maksimmasalski, 2020-03-30:
@stephanosio Thank you for your explanation and review.

maksimmasalski, 2020-03-30:
@stephanosio Do you know why I receive that error message trying to write 0xDEADBEEF in ARM?
`"mov r12, 0xDEADBEEF;\n\t"`
/tmp/cc2aCrG2.s: Assembler messages:
/tmp/cc2aCrG2.s:738: Error: invalid constant (ffffffffdeadbeef) after fixup

stephanosio, 2020-03-30:
> @stephanosio Do you know why I receive that error message trying to write 0xDEADBEEF in ARM?
> `"mov r12, 0xDEADBEEF;\n\t"`
> /tmp/cc2aCrG2.s: Assembler messages:
> /tmp/cc2aCrG2.s:738: Error: invalid constant (ffffffffdeadbeef) after fixup

You cannot specify 32-bit immediate to `mov` instruction; the best option you have is to load the value from the literal pool.

Try `ldr r12, =0xDEADBEEF`.

maksimmasalski, 2020-03-30:
@stephanosio Thanks.
Do you know how that x86 assembly command transfer into ARC assembly inline code?

stephanosio, 2020-03-30:
> @stephanosio Thanks.
> Do you know how that x86 assembly command transfer into ARC assembly inline code?

Not really an expert on the ARC stuff, but it seems 32-bit immediates (limm) are supported by `mov`.

As for the regs, see https://github.com/zephyrproject-rtos/zephyr/blob/386e5ba81c4bdfe782567fe380d77bee0ede871b/arch/arc/include/kernel_arch_data.h

maksimmasalski, 2020-04-03:
@stephanosio I want to use array to use instead of assigning for each variable register value.
But I noticed that my new implementation with array has final value which is not the same if to use just a simple variable to store register value, for example %ebx
**Old approach**
```
__asm__ volatile (                          
        "\t movl %%ebx,%0" : "=r"(ebx_val));
printk("\nReg ebx val user mode 0x%x\n", ebx_val);
```
Output: Reg ebx val user mode 0x11108c

**New approach:**
```
__asm__ volatile (
        "\t movl %%ebx,%0" : "=r"(x86_reg_val[0]));
printk("\nReg %s val user mode 0x%lx\n", x86_64_reg_name + i, 
        x86_64_reg_val[i]);      
```
Output: Reg ebx val user mode 0x12ff98

I find out that approach how C detects array and how asm detects array is not the same. What I'm doing wrong if instead of variable I'm using an array for storage register value?

closed 2020-09-28 by maksimmasalski