zephyrproject-rtos/zephyr #23877: syscall use of output buffers may be unsafe in some situations

andrewboie, 2020-03-27:
For syscalls that have output memory buffers, a lot of them currently do Z_SYSCALL_MEMORY_WRITE() on an out parameter before invoking the implementation function, and pass the out pointer directly to it.

This verifies that the caller has access to this memory, but may be unsafe. It should be safe if the output parameter is truly only used for output by the implementation. That is to say, this output buffer is ONLY written to, never read.

But if the system call ever uses this memory for internal working state, it's possible that there are preemption or SMP scenarios where this working state could get corrupted by some thread that has access to this memory, causing undefined behavior which at worst case could crash the kernel. I don't know if any current syscalls are vulnerable to this.

In addition, to properly implement in the future security features like SMAP, the kernel should never touch user memory at all except in designated copy_to_user()/copy_from_user() functions.

I think we need to audit our system calls that have out parameters and change verifiers that look like this:

```
int z_vrfy_some_syscall(int *out_param)
{
    Z_OOPS(Z_SYSCALL_MEMORY_WRITE(out_param, sizeof(*out_param)));
    return z_impl_some_syscall(out_param);
}
```

to

```
int z_vrfy_some_syscall(int *out_param)
{
    int local_out_param;
    int ret;
  
    ret = z_impl_some_syscall(&local_out_param);
	Z_OOPS(z_user_to_copy(out_param, &local_out_param, sizeof(*out_param)));
    return ret;
}
```

For syscalls that have output parameters that are integral types or fixed-size structs, this is straightforward.

We also need to consider syscalls where the end user passes in some buffer to be filled with data:

```
int z_vrfy_some_other_syscall(void *buf, size_t size)
{
    Z_OOPS(Z_SYSCALL_MEMORY_WRITE(buf, size));
    return z_impl_some_other_syscall(buf, size);
}
```

It doesn't perfectly feasible to do this with a stack allocation. I'm not completely sure how to proceed. I imagine we could do:

1. Use z_thread_malloc() to draw from the caller's resource pool, pass that buffer to the implementation function, and copy it. This is not desirable on an RTOS due to the need for heap allocation.
2. Keep code-as is, but clearly document that the contents of the buffer must never be read on the kernel side, only written to. I believe most or all syscalls that take output buffers are already safe.

ceolin, 2020-03-30:
@andrewboie there is another thing that we will need to define. `z_user_to_copy` and `z_user_from_copy` can return `EFAULT` and if it is necessary allocate memory in the handler we may need to deal with `ENOMEM` as well. So, the question here is, should we always `Z_OOPS()` these functions or we should propagate their errors ? The latter will be necessary to review all syscalls documentation.
Without look it too deep, I rather use Z_OOPS because I can't see what the caller can do for those errors.

andrewboie, 2020-03-31:
> @andrewboie there is another thing that we will need to define. `z_user_to_copy` and `z_user_from_copy` can return `EFAULT` and if it is necessary allocate memory in the handler we may need to deal [...]

@ceolin this is covered by our specification in https://docs.zephyrproject.org/latest/reference/usermode/syscalls.html#handler-function it is appropriate to `Z_OOPS()` when we deal with memory access errors, and return `-ENOMEM` when a heap allocation fails.

andrewboie, 2020-03-31:
With respect to parameters and how to deal with them, they fall into several categories.

For purposes of terminology, we define a buffer as:
* a parameter tuple which is a void * memory pointer and an associated size_t provided by the caller, or is otherwise a blob of memory of a known address and size.
* No logic in the kernel does anything with the buffer other than copy data back and forth. There is no validation or inspection of its contents.

1. Parameters that are passed by value are copied into registers when the syscall is made, there are no TOCTOU issues with these.

2. Input parameters that are passed by reference, typically structs. This includes linked data structures. These should be copied to a stack allocation and that allocation passed to the implementation. I believe most/all of our handlers are correct with respect to this, but it will be good to audit.

3. Output parameters used to pass back additional integral return values. Currently a lot of syscalls that have these look like this:

```
int z_vrfy_some_syscall(int *out_param)
{
    Z_OOPS(Z_SYSCALL_MEMORY_WRITE(out_param, sizeof(*out_param)));
    return z_impl_some_syscall(out_param);
}
```

This is safe if and only if the implementation treats this as a write-only pointer,  in which case this is classified as a buffer. But if the kernel ever makes any decisions based on its contents, such as a counter, there may be attacks possible because the value can be changed by userland at any time. It's impossible to know if the implementation does additional parameter validation in addition to what is done in the verification function. I think it is safest and desirable to convert all of these to use a stack allocation instead, which is then copied:

```
int z_vrfy_some_syscall(int *out_param)
{
    int local_out_param;
    int ret;
  
    ret = z_impl_some_syscall(&local_out_param);
	Z_OOPS(z_user_to_copy(out_param, &local_out_param, sizeof(*out_param)));
    return ret;
}
```

4. Parameters passed by reference which are both input/output. A common design is to pass in a pointer to a size_t which is a maximum allowable size, which is then updated by the implementation to reflect the actual number of bytes processed. For these I think we should also always use a stack allocation:

```
int z_vrfy_in_out_syscall(size_t *size_ptr)
{
    size_t size;
    int ret;
  
	Z_OOPS(z_user_from_copy(&size, size_ptr, sizeof(size));
    ret = z_impl_some_syscall(&size);
	*size_ptr = size;
    return ret;
}
```

Note that I do not use z_user_to_copy() after the implementation is called. We have already validated that the caller has access to the memory, I think it's sufficient to just do `*size_ptr = size`. This is predicated on the assumption that if the caller has access to memory at the beginning of the syscall, it will have such access through the lifetime of the system call.

5. Finally, buffers. Currently, these are passed from userland to the implementation function. This is either directly as one of the parameters to the syscall, or members of some struct passed to the syscall by reference. The policy on buffers is predicated on two major assumptions:

* If we check access permissions at the beginning of the syscall, the result of that check holds for the entirety of the call.
* The implementation makes no validation or logical decisions based on the buffer contents. This is important because userland can potentially change the contents of the buffer at any time to whatever data it wants.

If we keep this scheme, the latter point must be well-documented.

The alternative is that we adopt a policy of never passing user buffers to syscall implementations, but this has nontrivial costs:

* The verification function will have to allocate an equally sized buffer from kernel memory. This is either drawn from the stack using C VLAs or allocations from the callers resource pool. Drawing from the resource pool is a heap allocation which can potentially fail due to running out of memory, and this is heavily discouraged by MISRA-C Directive 4.12. VLAs require compiler support, depending on compiler could be implemented in terms of a heap, and OSes like Linux have been actively removing use of VLAs.
* There is also the performance overhead of making a copy of what might be a substantial chunk of data; for example instead of the driver copying once from MMIO, it will have to copy to the heap blob from MMIO first, and then copy from the heap blob to the buffer supplied by the user.

andrewboie, 2020-03-31:
I've sent a patch to update the docs for this in #23974

andrewboie, 2020-06-01:
The docs are up-to-date with the best practice. I think we can close this ticket.

closed 2020-06-01 by andrewboie