zephyrproject-rtos/zephyr #23887: drivers: modem: question: Should modem stack include headers to put into zephyr/include?

sslupsky, 2020-03-28:
Should the include headers for the modem stack (modem_cmd_handler.h, modem_context.h, modem_iface_uart.h, modem_socket.h) be placed in the Zephyr include directory?

The reason I ask is that I experienced problem when using an out-of-tree driver.  I did this because I am prototyping a modem driver for the Arduino MKR WAN.  To have the includes properly resolved I had to add a prefix (absolute or relative path) to the include statements of my driver:
```
#include "../../../zephyr/drivers/modem/<include file>"
```

jukkar, 2020-03-29:
Atm the headers are in `drivers/modem` as they are currently not a proper API and will be fluctuating in the future. There are some ideas floating around what to do with them but nothing concrete has come out of it yet. Best option for you atm, is to upstream your out-of-tree driver if possible.
cc: @mike-scott

sslupsky, 2020-03-29:
Thanks for the feedback.  The relative path works for now so I'll continue to use that.  I will need to pay attention to when this issue does change though.

Any thoughts on when the API will be finalized?

From what I have done so far I have a comment regarding the API.  The existing drivers are "socket" focused.  I am not sure yet if this approach makes sense for something like LoRaWAN which does not involve an IP stack.  In my case, since the MKR WAN does not use sockets it is not possible to offload sockets.  As my starting point, I used the ublox-sara-r4 driver and I encountered some difficulty compiling the modem stack when I removed the sockets offloading.  I've gotten past that point now but I had to learn a lot more about the modem stack than I thought I needed to know ;-)  Also, the documentation is a little thin, particularly modem_socket.h, so I spent a lot of time in the code trying to figure it out.  This isn't a criticism, just an observation to note in case anyone is tracking things left to do.

Another observation regarding modem_cmd_handler.  It appears find_cmd_match() assumes all the commands are unique.  I think there may be a problem when a command is a substring of another command.  For instance, I do not think the following is possible since "+ERR" is a substring of all the listed modem commands below and find_cmd_match() uses strncmp() to find a match:
```
	MODEM_CMD("+ERR", on_cmd_error, 0U, ""),
	MODEM_CMD("+ERR_PARAM", on_cmd_param_error, 0U, ""),
	MODEM_CMD("+ERR_BUSY", on_cmd_busy_error, 0U, ""),
	MODEM_CMD("+ERR_PARAM_OVERFLOW", on_cmd_overflow_error, 0U, ""),
	MODEM_CMD("+ERR_NO_NETWORK", on_cmd_network_error, 0U, ""),
	MODEM_CMD("+ERR_RX", on_cmd_rx_error, 0U, ""),
	MODEM_CMD("+ERR_UNKNOWN", on_cmd_unknown_error, 0U, ""),
```

sslupsky, 2020-04-02:
Regarding the second issue about substrings, I created a PR to address this.
https://github.com/zephyrproject-rtos/zephyr/pull/24043

closed 2020-07-15 by github-actions[bot]

slide333333, 2022-11-08:
Are there any plans for this one?

I'm currently in the need to use the PPP driver and handle things like SMS and GNSS via GSM MUX additionally. But since there is no public access to the modem_context `iface` member, which does expose the AT channel, this is not possible.

Is including the modem_context.h via "../../../zephyr/drivers/modem/<include file>" still the only solution atm or am I missing something?