zephyrproject-rtos/zephyr #23893: server to client ble coms: two characteristics with notifications failing to notify the right characteristics at the client

StefJar, 2020-03-29:
Simple setup:

```
/* service org.openfencing.sensor uuid128=0xbf202546424fdc143d352c45a683b90c*/
struct bt_uuid_128 bt_uuid_org_openfencing_sensor = BT_UUID_INIT_128(0xBF, 0x20, 0x25, 0x46, 0x42, 0x4F, 0xDC, 0x14, 0x3D, 0x35, 0x2C, 0x45, 0xA6, 0x83, 0xB9, 0x0C);
/* characteristic org.openfencing.sensor.start uuid128=0xadd90206e4042b8a3969db3101180322*/
struct bt_uuid_128 bt_uuid_org_openfencing_sensor_start = BT_UUID_INIT_128(0xAD, 0xD9, 0x02, 0x06, 0xE4, 0x04, 0x2B, 0x8A, 0x39, 0x69, 0xDB, 0x31, 0x01, 0x18, 0x03, 0x22);
/* characteristic org.openfencing.sensor.event uuid128=0x06d9f3f2cad165cfd866be3cf42bb728*/
struct bt_uuid_128 bt_uuid_org_openfencing_sensor_event = BT_UUID_INIT_128(0x06, 0xD9, 0xF3, 0xF2, 0xCA, 0xD1, 0x65, 0xCF, 0xD8, 0x66, 0xBE, 0x3C, 0xF4, 0x2B, 0xB7, 0x28);
/* characteristic org.openfencing.sensor.status uuid128=0x7ee17ac4458eb593195a2e9f2a30d0d5*/
struct bt_uuid_128 bt_uuid_org_openfencing_sensor_status = BT_UUID_INIT_128(0x7E, 0xE1, 0x7A, 0xC4, 0x45, 0x8E, 0xB5, 0x93, 0x19, 0x5A, 0x2E, 0x9F, 0x2A, 0x30, 0xD0, 0xD5);

static u8_t fs_gatt_start[FS_CHAR_SIZE_START];
static u8_t fs_gatt_status[FS_CHAR_SIZE_STATUS];
static u8_t fs_gatt_event[FS_CHAR_SIZE_EVENT];

enum fs_gatt_attrb_indxs {
	fs_gatt_attrb_indx_primServ = 0,
	fs_gatt_attrb_indx_start = 1,
	fs_gatt_attrb_indx_event = 2,
	fs_gatt_attrb_indx_event_ccc = 3,
	fs_gatt_attrb_indx_status = 4,
	fs_gatt_attrb_indx_status_ccc = 5,
};

/* fencing service */
BT_GATT_SERVICE_DEFINE(fs_cvs,
	BT_GATT_PRIMARY_SERVICE(&bt_uuid_org_openfencing_sensor),

	BT_GATT_CHARACTERISTIC(&bt_uuid_org_openfencing_sensor_start.uuid,
			BT_GATT_CHRC_WRITE,
			BT_GATT_PERM_WRITE,
			NULL, write_fs_start, fs_gatt_start),

	BT_GATT_CHARACTERISTIC(&bt_uuid_org_openfencing_sensor_event.uuid,
			BT_GATT_CHRC_READ | BT_GATT_CHRC_NOTIFY,
			BT_GATT_PERM_READ,
			read_fs_event, NULL, NULL),
	BT_GATT_CCC(event_ccc_event_changed, BT_GATT_PERM_READ | BT_GATT_PERM_WRITE),

	BT_GATT_CHARACTERISTIC(&bt_uuid_org_openfencing_sensor_status.uuid,
			BT_GATT_CHRC_READ | BT_GATT_CHRC_NOTIFY,
			BT_GATT_PERM_READ,
			read_fs_currentStatus, NULL, NULL),
	BT_GATT_CCC(event_ccc_currentStatus_changed, BT_GATT_PERM_READ | BT_GATT_PERM_WRITE),
);

void myTest1(void) {
fs_gatt_status[0] = 0x2;
bt_gatt_notify(NULL, &fs_cvs.attrs[fs_gatt_attrb_indx_status], fs_gatt_status, sizeof(fs_gatt_status));
}

void myTest2(void) {
bt_gatt_notify(NULL, &fs_cvs.attrs[fs_gatt_attrb_indx_event], fs_gatt_event, sizeof(fs_gatt_event));
}
```
no program crashes errors while running test 1 or test 2

If I am checking the client gatt db via (nrf connect and own ble python script) I see that the status is written to the event gatt db characteristic. for me looks like that the characteristic handle/index is wrongly calculated.

my hardware setup: nRF52840 + nRF52840 Dongle

jhedberg, 2020-04-07:
The `BT_GATT_CHARACTERISTIC()` macro expands into two attributes, so my first guess is that your indices are wrong.

joerchan, 2020-04-07:
Count 2 per characteristic.
So that would explain why start notify, turns into event notify.
```
enum fs_gatt_attrb_indxs {
	fs_gatt_attrb_indx_primServ = 0,
        /* decl  1 */
	fs_gatt_attrb_indx_start = 2,
        /* decl  3 */
	fs_gatt_attrb_indx_event = 4,
	fs_gatt_attrb_indx_event_ccc = 5,
        /* decl  6 */
	fs_gatt_attrb_indx_status = 7,
	fs_gatt_attrb_indx_status_ccc = 8,
};
```

StefJar, 2020-04-07:
thx to @jhedberg and @joerchan the enum values were wrong
Maybe put some note on the gatt database generation into [GATT API](https://docs.zephyrproject.org/latest/reference/bluetooth/gatt.html) to avoid others tapping into that trap

As well all samples(so far) only having one characteristic per service that offers notifications/indications. p.e [cts service](https://github.com/zephyrproject-rtos/zephyr/blob/master/samples/bluetooth/peripheral/src/cts.c)

closed 2020-04-07 by StefJar

AndreKR, 2022-07-09:
> The BT_GATT_CHARACTERISTIC() macro expands into two attributes, so my first guess is that your indices are wrong.

OMG, those "helper macros" are already weird in general but how would anyone know *this*?

joerchan, 2022-07-18:
> OMG, those "helper macros" are already weird in general but how would anyone know this?

100% agree, CC: @alwa-nordic @jori-nordic This is a common gotcha that creates problems for many users of the API. This could be improved in documentation, which is very sparse when it comes to how to create and handle a GATT database. The samples only demonstrate simple ones.

alwa-nordic, 2022-07-18:
For now, please use `bt_gatt_find_by_uuid` for locating the attribute in less fragile manner.